esphome/esphome-core #119: IR receive/transmit issue

definitio, 2018-08-01:
Hello,
I have problems with esphomeyaml on ESP8266:
1. ESP8266 reboots immediately after turning on remote switch.
Part of config:
```
remote_receiver:
  pin: 14
  dump: all

remote_transmitter:
  pin: 4
  carrier_duty_percent: 50%

switch:
  - platform: remote_transmitter
    name: "TV"
    panasonic:
      address: 0x4004
      command: 0x1000BCD
```
Log: [log.txt](https://github.com/OttoWinter/esphomelib/files/2250118/log.txt)

2. With receiver enabled it only works for a few seconds before reboot and shows a lot of raw codes even without pressing buttons on ir remote.

It works with examples from IRremoteESP8266 so the issue isn't hardware related.

definitio, 2018-08-01:
Fixed reboots for transmitter by disabling web interface. Receiver still causes restarts. IR switch doesn't work, devices don't recognize commands.

OttoWinter, 2018-09-27:
Should be fixed in 1.8.0

closed 2018-09-27 by OttoWinter