esphome/esphome-core #120: [Feature request] Interrupts

lobradov, 2018-08-07:
It would be good to have Interrupt driven GPIO for input (both directly connected or via I2C)
That would remove constant polling and would reduce time to response to events like buttons.

Any pointers as to where to start messing with the code?

OttoWinter, 2018-08-12:
So even if esphomelib would support getting the GPIO binary state via interrupts *for binary sensors*, the state changes could only be sent via MQTT on a polling basis. The reason is that interrupts are really only designed to do some very basic (and quick) stuff, but doing longer stuff like sending values over WiFi while in an ISR context is not possible.

The operation with interrupts would therefore need to be that an interrupt writes to some memory address that the GPIO state has changed. But esphomelib would still need to poll that memory address for updates, therefore defeating the purpose of interrupts. Anyway, I think the response time for buttons is quite low anyway.

At least on my Sonoff S20 a button press results in a seemingly instant response of the relay using esphomelib's existing logic. (Tip: In the auto-generated `main.cpp` file, there's a line `delay(16);` in the `loop()` function, you can decrease that to have the loop execute more often.)

lobradov, 2018-08-13:
It might not be a big deal on directly connected GPIO (quick to poll), but think multiple I2C extenders providing IO - instead of polling them all all the time, you could just poll the one that raised interrupt, only when it raised the interrupt.  That is where I expect most of the "saving", since each I2C operations is quite time consuming. 

I agree about MQTT and all - task manager with different task priorities would probably help there, but that might prove to be overkill for the job.

OttoWinter, 2018-09-19:
I mean I get your point, of course that would be more efficient, but I also think even polling i2c is not *too* time consuming. Please correct me if I'm wrong but even polling a sensor 8 times per `loop()` should not affect the default run rate of 60 loops per second (caused by the `delay(16);` in the main.cpp)

lobradov, 2018-10-08:
You are probably right - I'll close for the time being and see how it goes. If it's not good, I'll reopen and we can start playing with interrupts.

closed 2018-10-08 by lobradov