esphome/esphome-core #121: Platform for Wemos ESP-WROOM-02

ezar, 2018-08-08:
I have a [https://www.ebay.com/itm/WeMos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-Nodemcu-Module-18650-Battery-/172684188367](wemos) and I want to manage using esphomelib. What platform we need to use?
Regards.

lobradov, 2018-08-08:
http://docs.platformio.org/en/latest/platforms/espressif8266.html#boards

`esp_wroom_02`

davericher, 2018-09-07:
it is actually lolin32

esphomeyaml:
  name:                  aThing
  platform:              ESP32
  board:                 lolin32

OttoWinter, 2018-11-27:
In esphomeyaml 1.10.0 the user will be shown a dialog in the dashboard with all available platform options. And instructions what to do when the platform is not found. Part of https://github.com/OttoWinter/esphomeyaml/pull/234 (will be split up into multiple PRs)

closed 2018-11-27 by OttoWinter