esphome/esphome-core #126: Is there a timer functionality?

asantaga, 2018-08-10:
Hey Otto,

Im deploying a sensor of mine , currently using native arduino code but Im also looking at esphomelib as it would give me a bunch of things FOC (resiliance, remote logging etc).. My sensor has a binary sensor and a temperature sensor.. Currently works fine with HA, but Im wondering if its possible to send the status of all the sensors periodically regardless of change..  

how is this currently managed ? For example is a binary sensor only updated when it changes? or does it get updated reguarly?

Can I query the status programatically from HA?

many thanks, and please excuse me if ive missed something basic here.

OttoWinter, 2018-08-12:
Yes, binary sensors only send their state when the state changes. You can manually make the binary sensor publish its state by using the extended API like this (esphomeyaml [`lambda`](https://esphomelib.com/esphomeyaml/components/binary_sensor/index.html#lambda-calls)):

```cpp
id(my_binary_sensor).send_state_(id(my_binary_sensor).value);
```

Alternatively, you can use the [web server REST API](https://esphomelib.com/web-api/index.html) to query the status.

Sensors (like temperature) *do* send out values periodically even if the value hasn't changed.

asantaga, 2018-08-13:
Thanks Otto,  

I still dont see a "timer" perse.. I have two ideas

1. Use the on_message automation and listen out for a known message on my network.. When I receive it , I can then send the state.. perhaps listen out for the temperature sensor....
2. I could add "time " code the main loop

Perhaps on_timer is a good ER here?

OttoWinter, 2018-09-25:
@asantaga Ok, I'm not 100% I understand the question.

So: sensors (like the temperature sensor) already do send out their value periodically and regardless of change.

binary_sensors do only send out values on change, but since today in 1.8.0 there's now a `heartbeat` filter to manually send messages every so often. See #140.

closed 2018-09-25 by asantaga