esphome/esphome-core #128: DHT11 only working without checksum verification

janpieper, 2018-08-16:
I have a ESP-12E, a DHT11 (3 pins) sensor and the following configuration:

```
- sensor:
  - platform: dht
    model: DHT11
    pin: GPIO2
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
```

The wiring:

* (ESP-12E) GND - GND (DHT11)
* (ESP-12E) 3V3 - VCC (DHT11)
* (ESP-12E) D4 - DATA (DHT11)

When checking the logs, I see errors and warnings popping up because the checksum verification fails :disappointed: 

```
...
[08:18:38][E][sensor.dht:151]: Checksum invalid: 71!=80
[08:18:38][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[08:18:53][VV][sensor.dht:141]: Data: Hum=0b0010111100000000, Temp=0b0001100000001001, Checksum=0b01010000
[08:18:53][E][sensor.dht:151]: Checksum invalid: 71!=80
[08:18:53][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[08:19:08][VV][sensor.dht:141]: Data: Hum=0b0100000100000000, Temp=0b0001100100000100, Checksum=0b01011110
[08:19:08][E][sensor.dht:151]: Checksum invalid: 90!=94
[08:19:08][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[08:19:23][VV][sensor.dht:141]: Data: Hum=0b0100110000000000, Temp=0b0001101000000111, Checksum=0b01101101
[08:19:23][E][sensor.dht:151]: Checksum invalid: 102!=109
[08:19:23][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[08:19:38][VV][sensor.dht:141]: Data: Hum=0b0011011000000000, Temp=0b0001101100000011, Checksum=0b01010100
[08:19:38][E][sensor.dht:151]: Checksum invalid: 81!=84
[08:19:38][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
...
```

But, when I comment out the checksum verification...

https://github.com/OttoWinter/esphomelib/blob/1ee6b36aed1f006beeb671960fb44eebf10c9530/src/esphomelib/sensor/dht_component.cpp#L145-L156

... everything is fine:

```
[08:59:37][I][application:090]: Running through first loop()
[08:59:37][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=57.0%
[08:59:37][D][sensor.mqtt:091]: 'Temperature': Pushing out value 24.000000 with 1 decimals of accuracy
[08:59:37][D][sensor.mqtt:091]: 'Humidity': Pushing out value 57.000000 with 0 decimals of accuracy
[08:59:37][I][application:106]: First loop finished successfully!
[08:59:51][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:00:06][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:00:21][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:00:36][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:00:51][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:01:06][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:01:21][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
[09:01:30][I][ota:139]: Boot seems successful, resetting boot loop counter.
[09:01:36][D][sensor.dht:052]: Got Temperature=24.0°C Humidity=55.0%
```

How to get it working **without** monkey patching esphomelib?

bgoll, 2018-08-22:
+1

I have exactly the same problem and was able to (temporarily) fix it due to your hint. Thanks for that!

Would also realy like to see this fixed.

MattXcz, 2018-08-29:
Please fix this. DHT 11 has 0 decimals. That may be problem

pleasanthill17, 2018-09-09:
I have the same problem. @janpieper how can i change those lines in hassio addon. Where is that file located ?

janpieper, 2018-09-09:
You need to run `esphomeyaml <file> compile` first. It will download and compile the dependencies to be able to build the firmware you want to upload to your device. After that, you can find the file located in `<project-dir>/.piolibdeps/esphomelib/src/esphomelib/sensor/dht_component.cpp`, where `<project-dir>` is named like your YAML-file without the `.yml` or `.yaml` extension.

```
$ esphomeyaml example.yml compile
$ vi example/.piolibdeps/esphomelib/src/esphomelib/sensor/dht_component.cpp
```

Search for these lines...

```cpp
  uint8_t checksum;
  if (this->model_ == DHT_MODEL_DHT11)
    checksum = data[0] + data[2];
  else
    checksum = data[0] + data[1] + data[2] + data[3];

  if (checksum != data[4]) {
    if (report_errors) {
      ESP_LOGE(TAG, "Checksum invalid: %u!=%u", checksum, data[4]);
    }
    return false;
  }
```

... and comment them out:

```cpp
/*
  uint8_t checksum;
  if (this->model_ == DHT_MODEL_DHT11)
    checksum = data[0] + data[2];
  else
    checksum = data[0] + data[1] + data[2] + data[3];

  if (checksum != data[4]) {
    if (report_errors) {
      ESP_LOGE(TAG, "Checksum invalid: %u!=%u", checksum, data[4]);
    }
    return false;
  }
*/
```

Now you can re-compile the firmware and everything is fine.

Maybe I can find some time to have a deeper look at this problem and provide a PR. It's really annoying to do that monkey patching over and over again, when starting a new project.

MattXcz, 2018-09-10:
solution from @janpieper works, but it is not actually a solution...

janpieper, 2018-09-10:
:crossed_fingers: :crossed_fingers: #142 :crossed_fingers: :crossed_fingers:

pleasanthill17, 2018-09-14:
@janpieper thanks for the instructions to manually edit the file. That worked !!

OttoWinter, 2018-09-20:
FWIW, for developing the following is quite useful:

```bash
# in mynode.yaml, set library_uri in esphomeyaml: section to ''
cd mynode
mkdir lib
ln -s /my/dir/where/I/cloned/esphomelib lib/esphomelib
```

Then you can develop in your own copy of esphomelib. Anyway, apparently my source for the DHT11 checksum was only partially correct. For my DHT11, the old version works correctly but your change breaks things.

So the DHT22 use the sum of all 4 data bits as a checksum, and my DHT11 only uses only the 1st and 3rd. Your DHT11 apparently uses the DHT22's algorithm. Weird.

I will write a small fix which, if the DHT11 model is selected will allow either one of the checksums to pass.

janpieper, 2018-09-20:
Okay, that sounds weird. But as I said, I haven't tested my changed implementation with any other sensor (DHT21, DHT22). Here are two photos of one of the DHT11 sensors I use:

[image] [image]

OttoWinter, 2018-09-27:
Fixed in 1.8.0 - Please reopen if it still doesn't work

closed 2018-09-27 by OttoWinter