esphome/esphome-core #131: generic TCP and UDP server component

escoand, 2018-08-28:
I would like to request the ability for creating a generic TCP and UDP server. So we could communicate with the ESP without the need for a full fledged WebServer.

The transmitted data/protocol could be as low or high-level as the user wants. With this we could have more throughput and less overhead for many different use cases.

In my mind is the library binding the port and waiting for incoming connections and transmissions. The user code can create callback functions like the webserver api currently does.

The webserver at the end could be one implementation of this generic component. I think other more or less high-level protocols could follow.

OttoWinter, 2018-09-25:
I mean in principle, yes. But I do doubt that anybody would like to write raw HTTP messages.

The other thing is that if this component should really be as general as you propose, why not use the `WiFiServer` class in the Arduino framework directly. Or am I misunderstanding this a bit?

escoand, 2018-10-14:
Yes, this seems to be the right hint. But this would be out of scope of esphomeyaml, right?