esphome/esphome-core #132: Wrong pressure values using BMP180 and ESP32

Esiravegna, 2018-08-28:
After a couple of days testing the sensor, I got values which are about 1/3 of my current pressure:

```
[20:13:20][D][sensor.bmp085:133]: Got Pressure=325.6hPa
```
vs:

```
Presión: 958.1hPa
```

However, not is only the temperature fine, but it also matches my DHT22 on the same board.

I've tried even creating a custom component based on AdaFruit085 as opposed to the built int component, with similar results. I even used another BMP180 (both of them are GY-46 based), and a different ESP32 (NodeMCUESP32s vs a DoIt DevKit). All the same results.

I also tried compensating the altitude via using a formula:

```
      - lambda: >-
          return (x / pow(1.0-(477/44330), 5.255));
```
(My altitude is 477 meters above sea level), to no avail.

Any ideas would be very appreciated.
Thans, and THANKS A LOT for all your hard work in this!

Esiravegna, 2018-08-30:
I got some closer to the actual reading results by using [this library](https://github.com/LowPowerLab/SFE_BMP180/blob/master/examples/SFE_BMP180_example/SFE_BMP180_example.ino)

OttoWinter, 2018-09-19:
Hmmm, that's really weird. I just checked my BMP180 again and there it seems to give good pressure values. Could you turn on the `very_verbose` logs in the `logger` component and give me those logs? They include all the data sent to the BMP180 and back and should help resolve the issue.

OttoWinter, 2018-09-27:
Could you check with v1.8.0? I think it could be fixed there.

Esiravegna, 2018-09-27:
Sure, thanks a lot for your work.

Esiravegna, 2018-10-17:
Same values:
```
[18:44:59][D][sensor.bmp085:135]: Got Pressure=318.8hPa
[18:45:14][D][sensor.bmp085:089]: Got Temperature=26.6┬░C
[18:45:14][D][sensor.bmp085:135]: Got Pressure=318.8hPa
[18:45:29][D][sensor.bmp085:089]: Got Temperature=26.6┬░C
[18:45:29][D][sensor.bmp085:135]: Got Pressure=318.8hPa
[18:45:30][D][binary_sensor.mqtt:070]: 'Tower Radar': Sending
```

Esiravegna, 2018-10-17:
very_verbose logs:
```
[18:51:20][VV][component:136]: Running interval 'update':0 with interval=15000 last_execution=59061 (now=74065)
[18:51:20][V][sensor.bmp085:147]: Setting mode to 0x2E...
[18:51:20][VV][i2c:079]: Beginning Transmission to 0x77:
[18:51:20][VV][i2c:117]:     Writing 0b11110100 (0xF4)
[18:51:20][VV][i2c:117]:     Writing 0b00101110 (0x2E)
[18:51:20][VV][i2c:084]:     Transmission ended. Status code: 0x00
[18:51:20][V][component:081]: set_timeout(name='temperature', timeout=5)
[18:51:20][VV][component:136]: Running timeout 'temperature':1 with interval=5 last_execution=74098 (now=74115)
[18:51:20][VV][i2c:079]: Beginning Transmission to 0x77:
[18:51:20][VV][i2c:117]:     Writing 0b11110110 (0xF6)
[18:51:20][VV][i2c:084]:     Transmission ended. Status code: 0x00
[18:51:20][VV][i2c:106]: Requesting 2 bytes from 0x77:
[18:51:20][VV][i2c:136]:     Received 0b01110000 (0x70)
[18:51:20][VV][i2c:136]:     Received 0b00010100 (0x14)
[18:51:20][D][sensor.bmp085:089]: Got Temperature=26.6┬░C
[18:51:20][V][sensor.sensor:026]: 'Outside Temperature': Received new value 26.600000
[18:51:20][V][sensor.bmp085:147]: Setting mode to 0xF4...
[18:51:20][VV][i2c:079]: Beginning Transmission to 0x77:
[18:51:20][VV][i2c:117]:     Writing 0b11110100 (0xF4)
[18:51:20][VV][i2c:117]:     Writing 0b11110100 (0xF4)
[18:51:20][VV][i2c:084]:     Transmission ended. Status code: 0x00
[18:51:20][V][component:081]: set_timeout(name='pressure', timeout=26)
[18:51:20][VV][component:136]: Running timeout 'pressure':1 with interval=26 last_execution=74191 (now=74223)
[18:51:20][VV][i2c:079]: Beginning Transmission to 0x77:
[18:51:20][VV][i2c:117]:     Writing 0b11110110 (0xF6)
[18:51:20][VV][i2c:084]:     Transmission ended. Status code: 0x00
[18:51:20][VV][i2c:106]: Requesting 3 bytes from 0x77:
[18:51:20][VV][i2c:136]:     Received 0b10010101 (0x95)
[18:51:20][VV][i2c:136]:     Received 0b10011101 (0x9D)
[18:51:20][VV][i2c:136]:     Received 0b01100000 (0x60)
[18:51:20][D][sensor.bmp085:135]: Got Pressure=318.8hPa
[18:51:20][V][sensor.sensor:026]: 'Outside Pressure BMP': Received new value 318.779999
[18:51:21][VV][component:136]: Running interval 'update':0 with interval=15000 last_execution=59347 (now=74352)
[18:51:21][V][sensor.sensor:026]: 'Outdoor Light Level': Received new value 4.860000
```

PR #292 (2018-12-01): Fix BMP085 pressure value
## Description:

**Related issue (if applicable):** fixes #132

## Checklist:
  - [x] The code change is tested and works locally.
  - [x] Check this box if you have read, understand, comply, and agree with the [Code of Conduct](https://github.com/OttoWinter/esphomelib/blob/master/CODE_OF_CONDUCT.md).
closed 2018-12-01 by OttoWinter