esphome/esphome-core #134: Wemos Shut Down Tasmota

francylife89, 2018-09-01:
Dear everyone,

I'm new and I desperately need your help. I'm using 8 wemos d1 mini with Tasmota software version 5.14.0. I installed the devices a few months ago with mqtt disabled (I had no internet connection and hassio impaled) to control the lights at home and everything seemed to work perfectly until ... I connected everything to the internet and hassio, so I enabled 'mqtt and suddenly the wemos are no longer reachable (randomly, not all at once) via the web server and I can not ping them. I reported everything to the initial configuration but the problem persists. Although they are not reachable I can turn on the lights from the buttons. The only way to reach them via the network is to press the reboot button on the wemos.

Does anyone have any idea why this thing happens? Please help me

mverleun, 2018-09-03:
You're question is outside the scope of esphomelib. You should ask it at the Tasmota github site.

But some of the problems can be experienced with any platform using MQTT.
Assuming that the network configuration is fine (DHCP) and no IP address conflicts exist it is quite likely that you're clients are all using the same client id when connecting to the MQTT broker. 
If this is the case the broker get's confused, it sees multiple IP addresses being assigned to a client id and will only work with the last IP address it sees. All other IP addresses are ignored.

This behaviour is normal, it is quite common for MQTT devices to move between networks with changing IP addresses as a result. Hence the MQTT broker can only use the client id as a reliable reference. Therefor this client id has to be unique.

I'm not a Tasmota expert but can imagine that if the MQTT connection keeps getting dropped frequently because of this issue it will cause the devices to restart in a attempt to recover from this situation.

francylife89, 2018-09-04:
Dear mverleun,
thank you very much for your suggestion. I checked the ID clients and, although they were all different for the various devices, from the MQTT logs I noticed that some devices were logging in with the same ID. So I changed all the client IDs and right now everything seems working.
Thanks for your support you have been a great help.

closed 2018-09-14 by francylife89