esphome/esphome-core #135: Add more IR remote codes

escoand, 2018-09-02:
I was looking for a way to send and receive IR codes to and from my Samsung tv. Currently esphomelib is not able to handle them but just the raw telegram.

But I found a description of the codes for  Samsung and even more manufacturer.
I can confirm the Samsung codes are as described for my concrete device.

Could they get implemented and maybe in a more simple way than the current one-manufacture-one-class way. Maybe as a simple config file.

OttoWinter, 2018-09-20:
Yes for encoding it's mostly that simple. However when it comes to *decoding* it can quickly get more complicated. For example with sony codes only 12 bit, 15 bit or 20 bit codes are valid.

Anyway, I think it *would* be possible to create a generic IR code class, but I don't really have a problem with the existing solution of having one class per manufacturer. It's a bit more difficult to maintain, but ultimately it saves me from creating a monstrous class that attempts to support every single IR code there is.

yawor, 2018-11-18:
@OttoWinter I'd add another aspect to this issue. Naming of the supported IR codes (LG, Samsung, Sony etc) is far from good. There's no guarantee at all that a single brand is always going to use the same IR protocol. For example I have LG TV and Soundbar and the TV uses NEC1, but the Soundbar uses NECx2. In esphomelib, NEC2/NECx2 is implemented as samsung.
I think esphomelib should use protocol names instead of equipment brands. So samsung should probably be renamed to nec2, nec should be renamed to nec1. Sony doesn't need renaming as it's actually Sony's own protocol and is referred to as such (usually as sony8, sony12, sony15, sony20 depending on the frame length). Panasonic is also ok, as it's the same situation as with sony, but without variants. I don't recognise the protocol defined in the lg.cpp - seems similar to NEC1, but it's off a little.

@escoand it's not that easy. All IR protocols implemented right now in esphomelib (nec1, nec2, sony, panasonic) use a burst/gap timing encoding. This is quite simple to decode as you say, as the rising and falling edges are always within a single bit. But there are also phase-shift encoding schemes, where the order of high/low signal define 0 and 1 (for example high/low -> 0, low/high -> 1). For example this means that if you have two bits 00, then the signal is going to look high/low/high/low, but if you have bits 01 then the signal would look like high/low*2/high (low*2 means keep the signal low twice long as normal) so you don't have a rising or falling edge between bits. It's of course still possible to decode but it's a bit harder than a simple gap protocol.
There are also protocols where multiple bits are encoded in a single burst/gap pair, which has multiple possible timings (for example 00 -> 600us/600us, 01 -> 600us/800us, 10 -> 600us/1000us and 11 -> 600us/1200us). An extreme example of such protocol is the XMP, which encodes 4 bits in a single pair with 16 different gap times (from 760us up to 2800us with 136us increments).

yawor, 2018-11-18:
As a reference for different IR signals and protocols, I can recommend IrScrutinizer program. It's written in Java so it can be run on any desktop system. You can use Generate tab in the program to generate an IR signal for any IR protocol added in the program (and there's a lot). With proper hardware it can also send and receive IR signals. It can be used to analyse and decode signals.

escoand, 2018-11-18:
OK, you seem to have much more experience with all that. My knowledge was just some googling and try and catch.
I also tried to reverse engineer my Logitech remote but this seems to be some of your described phase-shift protocols.
Good would be a existing collection which could be used here.

yawor, 2018-11-18:
@escoand download the program I've recommended. It's open source and is being developed by a member of hifi-remote (jp1) community. There's a really big collection of IR protocols included in the program. You just select one from the dropdown, enter some values and you can generate a signal for provided parameters.
If you mean Logitech Harmony remote, then the protocol it uses depends on the devices configured on the remote. But there's also a logitech proprietary protocol, but it's not phase-shift, but a simple gap timing.

OttoWinter, 2018-11-18:
The names of the IR protocols are based on https://github.com/markszabo/IRremoteESP8266/ and personally I don't really mind these protocol names not being 100% correct. Most users who have previously used IR using another firmware probably did so directly or indirectly through the IRremote library. So if you already have the IR codes for the other frameworks you can use them here too.

Also, when "dumping" these remote codes using the remote receiver component, you get the name of the protocol too. So when the user presses the remote on an LG soundbar, they get a message saying "samsung code=bla ..." - and the user probably doesn't really care whether it says lg or samsung as long as it works :)

The other aspect to this is that renaming the protocol would be a breaking change, and I don't like breaking changes unless they're absolutely necessary. And even if the `lg`, ... protocol names are a bit off, they're far more memorable than the `nec1`, `nec2`, ...

I mean I don't really care that much. Maybe if I would re-do this implementation I would change it, but personally I don't really see a big reason why to change it now. Yes it's technically incorrect but as long as it works ¯\_(ツ)_/¯

yawor, 2018-11-18:
@OttoWinter that's fair enough. I've been thinking about adding some form of aliases to keep compatibility with the old names, but it's probably not worth doing that. At least not when there are a lot of other issues and features to be taken care of.