esphome/esphome-core #137: Ethernet Shield Support

Odianosen25, 2018-09-03:
Hello,

Many thanks for your great system. Please if I may ask, is it possible to use your code with an Ethernet shield? If it’s possible, can you kindly point to the right one?

Is it so straight forward, or there are other things to be done?

Thanks and regards

OttoWinter, 2018-09-20:
Currently not. Though esphomelib is made specifically for *ESPs* (not for generic arduinos!) which pretty much per definition have WiFi, so I don't really understand why ethernet shields would be useful... What is your use-case.

Odianosen25, 2018-09-20:
Ok, and many thanks for the response.

I kind of mix the ESP chip with Arduinos. Basically I wanted to be able to connect a LAN cable to an ESP chip, and make use of your library to turn them into sensors. So the data can be sent over MQTT to a remote broker. This way I can do away with the WIFI for extra reliability.

Regards

OttoWinter, 2018-09-25:
Ok, that sounds good. What ethernet shield were you thinking about/what is the most common one?

Odianosen25, 2018-09-28:
Hello @OttoWinter,

Apologies I am just responding, I totally forgot after reading the mail.

I am looking at possibly this one [here](https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware).

Regards

readylan, 2018-10-10:
Hello guys, I am very interested in that thing too. My use-case is for esp32/esp8266 based sensors (wired and not wired) creating a network where wired ones can do the "gateway role". The use of ethernet with those "gateway nodes" is for reliability ([and powering with boards like this one](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware) from the same manufacturer)

Thanks and regards

OttoWinter, 2018-10-10:
I looked at this again, and it seems much more complicated to do than I thought.

The whole MQTT client and web server component would basically need to be rewritten to support ethernet shields/boards. The reason is that the APIs for ETH and WiFiClient are quite different and the libraries used for MQTT and the web server both only work via the WiFi API.

Sorry, I won't do this any time soon.

@readylan So you want these "gateway" ESPs to have a wired ethernet connection and have them be a WiFi access point at the same time? Sorry, I've looked at that before but it's very difficult to do and would only cause issues for esphomelib in "normal" operation. There are projects like [this one](https://github.com/martin-ger/esp_wifi_repeater) which you can use for that. But I do not plan on implementing that for esphomelib.

Odianosen25, 2018-10-10:
Many thanks for looking into it @OttoWinter, will keep looking into other devices that can possibly do it.

Kind regards 

You can close the thread if you want to :)

readylan, 2018-10-10:
Thank you very much and congratulations for the great job !

readylan, 2018-10-24:
I just read [this article](https://www.crowdsupply.com/silicognition/wesp32/updates/why-ethernet) about ESP32 interfaces and I totally agree with the author. We usually think in WiFi when is about ESPs, but ethernet can do it better in many scenarios.