esphome/esphome-core #138: MQTT command to toggle an component

escoand, 2018-09-03:
I would like to toggle a light of an other esphomelib device. This is currently just possible when subscribing to the state topic of the destination device but not with a toggle MQTT telegram.
What do you think about this additional command?

OttoWinter, 2018-09-20:
I mean for such use-case it would be pretty easy I believe to just use the [`on_message`](https://esphomelib.com/esphomeyaml/components/mqtt.html#on-message) trigger to do anything you want:

```yaml
# Node A
mqtt:
  on_message:
    topic: my/toggle/topic
    then:
      - switch.toggle:
          id: my_switch
```

```yaml
# Node B
on_...:
  # Toggle switch on other node
  mqtt.publish:
    topic: my/toggle/topic
    payload: TOGGLE
```

I do not really intend to add a specific toggle command to the MQTT frontend if it's not really too big of a use-case + it's possible in other ways.

escoand, 2018-09-20:
Thanks, I'd already solved it in that way.

For me toggling is as basic as on/off. Otherwise the sender always has to know the current state. For HA this is not an issue but for any other "dump" device it's a lot of overhead. Or the other way around, every switch has to add the toggle command by hand.

OttoWinter, 2018-09-25:
The reason I was hesitant at first was that I didn't want to make the nodes subscribe to a new "toggle" topic by default because each subscription can lead to more memory/network problems. But I totally forgot that the payload can also be used. See #152.

escoand, 2018-09-25:
Thanks man, good work!

PR #152 (2018-09-26): Add toggle payload to MQTT switch component
When payload "TOGGLE" is sent to command_topic, it will now toggle the state of the switch.

Fixes #138
closed 2018-09-25 by escoand