esphome/esphome-core #139: Option to set MQTT update interval?

mverleun, 2018-09-03:
Hi,
I like you're project a lot, especially with the yaml configuration option this framework is great for educational purposes.

Hence my question about MQTT timing. At the moment the logging shows the updates of sensors at regular intervals, configured with update_interval in the yaml file.
When a sensor is updated it does not automatically publish the (new) sensor value to the MQTT broker. The updates sent to the MQTT broker take several minutes which is a bit long for impatient students. 
Is there an option to shorten this interval?

mverleun, 2018-09-04:
This morning I figured out how to accomplish the requested behaviour.
It seems that, by default, MQTT updates are sent after 15 sensor readings have been done.
This was not clear to me, I was expecting a one on one relation by default.

Luckily it is possible to use filters to get this one on one behaviour.

OttoWinter, 2018-09-20:
Oh man, RTFM... :) no but really, for this exact reason every single `update_interval` option in the docs has a "See [Default Filter](https://esphomelib.com/esphomeyaml/components/sensor/index.html#sensor-default-filter)." link so that users can find out about this quicker. Closing...

closed 2018-09-20 by OttoWinter