esphome/esphome-core #143: Template Binary sensor causing loop

barryodonovan, 2018-09-11:
Template Binary Sensor is causing a loop during (mqtt?) setup on an ESP32 for me.

The code causing the issue is:
'sensor:
  - platform: adc
    pin: GPIO32
    name: "Pellet Hopper Light"
    id: pellethopperlight
    update_interval: 60s
    attenuation: 6db

binary_sensor:
  - platform: template
    name: "Pellet Hopper Level"
    device_class: problem
    lambda: >-
      if (id(pellethopperlight).value > 0.2) {
        return true;
      } else {
        return false;
      }
`

After complilation and uploading the logs show multiple recurring MQTT setup debug output,  on my frontend sensors flick to/from unavailable states. No values appear to be polled and none are returned.

OttoWinter, 2018-09-20:
Hmmm, I just compiled&run this code and it worked correctly. Anything more specific in the logs? What do you see if you enable verbose logs?

OttoWinter, 2018-11-27:
Closing this issue as I can't reproduce it...

closed 2018-11-27 by OttoWinter