esphome/esphome-core #148: DHT_Sensor setup

voicevon, 2018-09-19:
I am using DHT11 on NodeMcu-12.  in setup() function. there is only one line I copied from sample file.
App.make_dht_sensor("Outside Temperature", "Outside Humidity", D0);
When I launch it, I got the error as below

␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 16 with mode OUTPUT␛[0m

Exception (29):
epc1=0x401065b4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe40:  3ffffe80 3ffffe70 0000000c 3ffe8fe0
3ffffe50:  00000043 00000010 00003a98 40206928
3ffffe60:  3ffffe80 3ffffe70 0000000c 402055b0
3ffffe70:  3ffe8f34 3ffffe80 0000000c 3ffe9c70
3ffffe80:  00000021 00000000 00000004 402055d9
3ffffe90:  3ffffec4 00003a98 3fff3734 402091c1
3ffffea0:  00000006 00003a98 3fff3734 40204df4
3ffffeb0:  00000000 00000000 00000000 00000000
3ffffec0:  3fff1fa8 3fff2b54 3fff1148 4020ef16
3ffffed0:  00000005 3fff3734 3fff1148 40203ee6
3ffffee0:  3fff2014 00000006 3fff426c 402032e2
3ffffef0:  00000006 00000100 00000014 3fff3dbc
3fffff00:  3fffffa0 3fffff58 00003a98 3fff1558
3fffff10:  3fff2fdc 00003a98 3fff1148 4020b395
3fffff20:  40210170 3fff010c 60000308 60000304
3fffff30:  60000318 00001000 40210700 0100a8c0
3fffff40:  40210730 00ffffff 40210730 00000000
3fffff50:  40210730 00000000 3fff426c 3fff429c
3fffff60:  3fff42dc 0100a8c0 40210730 00ffffff
3fffff70:  40210730 00000000 40210730 00000000
3fffff80:  3fff3734 3fff3d14 3fff3d5c 3fff4174
3fffff90:  3fff41bc 3fff3dec 3fff40f4 3fff4240
3fffffa0:  feefeffe feefeffe feefeffe feefeffe
3fffffb0:  3fffdad0 00000000 3fff1550 4020eee0
3fffffc0:  feefeffe feefeffe 3ffe862c 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

----------------------------------------------------------------------------------------------------
I changed to pin D1.  same error as above.
-----------------------------------------------------------------------------------------------------
I changed to pin D2. sames much better. shows as 
␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 4 with mode OUTPUT␛[0m
␛[0;35m[C][component:243]:     Update interval: 15000ms␛[0m
...
␛[0;32m[I][application:093]: Running through first loop()␛[0m
␛[0;35m[C][sensor.dht:047]: Auto-detected model DHT22. Override with the 'model:' configuration option␛[0m
␛[0;32m[I][application:109]: First loop finished successfully!␛[0m
␛[0;33m[W][sensor.dht:120]: Falling edge for bit -1 failed!␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m

But actually I connected nothing to Pin D2, it;s empty!   After connected Pin D2 to DHT11 Pin Data, same information, nothing is changed. I don't think my hardware sensor has any issue(just bought, I can't prove it works fine. But it should be OK).

kaiomatico, 2018-09-21:
I also have problems with using a DHT-22 and esphomelib. I connected my DHT-22 to pin D2 and called this function: > App.make_dht_sensor("Schlafzimmer Temperature", "Schlafzimmer Humidity", D2);

But I get the following error log:
`[1;31m[E][sensor.dht:151]: Checksum invalid: 1!=204[0m
[0;33m[W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0[0m
`

The interesting thing is that it is working fine using esphomeyaml with this configuration:
`sensor:
  - platform: dht
    pin: D2`

So I guess the pin "D2" is not really translated correctly to the proper gpio pin on the esp8266?

OttoWinter, 2018-09-25:
Could you try with esphomelib v1.8.0? There were some timing issues before with the DHT integration but that is fixed now.

voicevon, 2018-09-27:
Hi kaiomatico and master, I tested V1.8.0.  It seems no help.

```
//Source code
  auto dht = App.make_dht_sensor("Temp", "Humi", D2);
  dht.dht->set_dht_model(sensor::DHT_MODEL_DHT11);

```

```
//Response
␛[0;32m[I][application:051]: Running through setup()...␛[0m
␛[0;35m[C][wifi:031]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:239]: WiFi Connecting to 'XXX'...␛[0m
␛[0;32m[I][wifi:290]: WiFi connected!␛[0m
...
␛[0;35m[C][mqtt.client:024]: Setting up MQTT...␛[0m
...
␛[0;32m[I][mqtt.client:099]: Connecting to MQTT...␛[0m
␛[0;32m[I][mqtt.client:135]: MQTT Connected!␛[0m
␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 4 with mode OUTPUT␛[0m
␛[0;35m[C][component:243]:     Update interval: 15000ms␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'Humi'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '%'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 0␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: 'mdi:water-percent'␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'Temp'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '°C'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 1␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: ''␛[0m
␛[0;32m[I][application:093]: Running through first loop()␛[0m
␛[1;31m[E][sensor.dht:153]: Checksum invalid: 81!=85␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[0;32m[I][application:109]: First loop finished successfully!␛[0m
␛[1;31m[E][sensor.dht:153]: Checksum invalid: 85!=90␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[1;31m[E][sensor.dht:153]: Checksum invalid: 81!=86␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[1;31m[E][sensor.dht:153]: Checksum invalid: 81!=86␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
```

And, I tested automatic find sensor.
```
//Source code
  auto dht = App.make_dht_sensor("Temp", "Humi", D2);
//  dht.dht->set_dht_model(sensor::DHT_MODEL_DHT11);

```

```
//Response
␛[0;32m[I][application:051]: Running through setup()...␛[0m
␛[0;35m[C][wifi:031]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:239]: WiFi Connecting to 'XXX'...␛[0m
␛[0;32m[I][wifi:290]: WiFi connected!␛[0m
...
␛[0;35m[C][mqtt.client:024]: Setting up MQTT...␛[0m
...
␛[0;32m[I][mqtt.client:099]: Connecting to MQTT...␛[0m
␛[0;32m[I][mqtt.client:135]: MQTT Connected!␛[0m
␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 4 with mode OUTPUT␛[0m
␛[0;35m[C][component:243]:     Update interval: 15000ms␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'Humi'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '%'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 0␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: 'mdi:water-percent'␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'Temp'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '°C'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 1␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: ''␛[0m
␛[0;32m[I][application:093]: Running through first loop()␛[0m
␛[0;35m[C][sensor.dht:047]: Auto-detected model DHT22. Override with the 'model:' configuration option␛[0m
␛[0;32m[I][application:109]: First loop finished successfully!␛[0m
␛[0;33m[W][sensor.dht:120]: Falling edge for bit -1 failed!␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[0;33m[W][sensor.dht:120]: Falling edge for bit -1 failed!␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[0;33m[W][sensor.dht:120]: Falling edge for bit -1 failed!␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
␛[0;33m[W][sensor.dht:120]: Falling edge for bit -1 failed!␛[0m
␛[0;33m[W][sensor.dht:061]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0␛[0m
//Repeating
```

BTW: I am buying DHT22, Will feedback  to all of you in few days.

OttoWinter, 2018-09-27:
@voicevon Are you sure you're on 1.8.0? Because the following line from your logs should only show up in 1.7.0

```
Falling edge for bit -1 failed!
```

I think the command to update is

```
platformio lib update
```

Or setting `esphomelib@1.8.0` in `lib_deps`

voicevon, 2018-09-30:
@OttoWinter  I was working with V1.7.0 on last comment. After I updated to V1.8.0. My DHT11 got the right response. But has some limitations with:
1) Must connect to Pin_D2 (GPIO4).  Can't work on Pin_D0(GPIO16), Pin_D1(GPIO5). So far I tested on these 3 pins, other pins has not been tested.
2) Don't use auto detecting DHT type(debugger still says found DHT22 and ). Must manually set_dht_model

```
   auto dht = App.make_dht_sensor("Temp", "Humi",D2);
   dht.dht->set_dht_model(sensor::DHT_MODEL_DHT11);

```

I am not sure whether should I put below message or to create one or more new topics.
1st)  In file platformio.ini,  add the following line code.Because the monitor_speed is 11500 by default  in esphomelib ,but it is 9600 in IDE.
 monitor_speed = 115200
2nd) after I put this line in file platform.ini     esphomelib@1.8.0 in lib_deps  .compiling got failure.

3rd) How can I know the version of esphomelib of my project?. Could I get it when compile the project?

4th) I am sure that after making a sensor object , the object has included some filters already, and I don't know what kind of filters, either what the parameters of those/that filters. What I have to do is to Clear filters manually with code like below, I hope the sensor has no filter after being constructed.

```
  auto dht = App.make_dht_sensor("Temp", "Humi",D2);
   dht.dht->set_dht_model(sensor::DHT_MODEL_DHT11);
   dht.dht->get_temperature_sensor()->clear_filters();
    dht.dht->get_humidity_sensor()->clear_filters();
```

voicevon, 2018-09-30:
One possible reason of D0 doesn't work is D0 is connected to an on board  LED , This is really happened on NodeMcu-12 .  Don't know the other boards.

voicevon, 2018-10-01:
My DHT-22 is arrived. I tested it. Same limitation as DHT 11.
The code must be:
```
   auto dht = App.make_dht_sensor("Temp", "Humi",D4);
   dht.dht->set_dht_model(sensor::DHT_MODEL_DHT22);   //can't remove this line.
```

OttoWinter, 2018-10-04:
@voicevon 9600 being the default baud rate in the IDE is the IDE's problem. Most ESPs have used the baud rate 115200 forever because it's the fastest for data transfer and doesn't cause any issues.

> after I put this line in file platform.ini esphomelib@1.8.0 in lib_deps .compiling got failure.

What failure? I can't help without logs.

> How can I know the version of esphomelib of my project? Could I get it when compile the project?

Yes, if you enable the debug component `make_debug_component` it will spit it out in the boot phase.

> 4th) I am sure that after making a sensor object , the object has included some filters already, and I don't know what kind of filters, either what the parameters of those/that filters. What I have to do is to Clear filters manually with code like below, I hope the sensor has no filter after being constructed, If I want to add some filters , let me do it myself.

There has always been a trade-off here. On the one side, all sensor data is noisy and so usually it's good to filter the data. Now for most use-cases, I think this is a good idea since it's not really a problem to increase the poll-frequency of the sensor, but it *is* a bigger problem to send more data over WiFi (+ when using Home Assistant, it clutters the state machine).

Of course that means that you'll have to manually disable the filters but I think it's a good trade-off between default parameters and customizability.

> My DHT-22 is arrived. I tested it. Same limitation as DHT 11.

Ok, in some cases the auto-detection fails because some DHTs are not performing to spec. Unfortunately, I can't change anything about that and you'll have to add that line to help the auto-detection. Do you have any messages like `Auto-detected model DHT11` in the logs?

voicevon, 2018-10-14:
I am using vscode+Plarformio+esphomelib1.8.2.  No yaml(hassio can't be installed.)  DHT22 is connected to pin Esp8266-D2.

My source code:
```
void setup() {
 auto* log = App.init_log();
   log->set_baud_rate(115200);
   log->set_tx_buffer_size(512);
   //log->set_global_log_level(0);

 auto *wifi = App.init_wifi();
 // *all* parameters have to be declared in *exact order*, as defined in declared structures.
 // otherwise, compiler will complain with "sorry, unimplemented: non-trivial designated initializers not supported"
 wifi->set_sta(WiFiAp{
     .ssid = "xxx",
     .password = "xxx",
     .channel = 0
 });
 auto *mqtt = App.init_mqtt("m10.cloudmqtt.com",18601, "xxx", "xxx");
         mqtt->set_topic_prefix("MyTopic");
  App.make_status_led(D0);   //Build-in LED.
 
  auto dht = App.make_dht_sensor("temp", "humi",D2);
  App.setup()
}
```
And Response

```
�p`rn|��n␄␌b␌�␎l␏r���␂␌␄#␌�␎$�#cb␌␄��␂rl ␃��o�␃␛[0;32m[I][log_component:055]: Log initialized␛[0m
␛[0;32m[I][application:051]: Running through setup()...␛[0m
␛[0;35m[C][status_led:026]: Setting up Status LED...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 16 with mode OUTPUT␛[0m
␛[0;35m[C][wifi:031]: Setting up WiFi...␛[0m
           ......
␛[0;35m[C][mqtt.client:024]: Setting up MQTT...␛[0m
      .....
␛[0;32m[I][mqtt.client:101]: Connecting to MQTT...␛[0m
␛[0;32m[I][mqtt.client:137]: MQTT Connected!␛[0m
␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 4 with mode OUTPUT␛[0m
␛[0;35m[C][component:243]:     Update interval: 15000ms␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'humi'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '%'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 0␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: 'mdi:water-percent'␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'temp'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '°C'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 1␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: ''␛[0m
␛[0;32m[I][application:088]: You're running esphomelib v1.8.2␛[0m
␛[0;32m[I][application:099]: Running through first loop()␛[0m
␛[0;35m[C][sensor.dht:045]: Auto-detected model DHT11. Override with the 'model:' configuration option␛[0m
␛[0;32m[I][application:115]: First loop finished successfully!␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.mqtt:094]: 'temp': Pushing out value 0.000000 with 1 decimals of accuracy␛[0m
␛[0;36m[D][sensor.mqtt:094]: 'humi': Pushing out value 2.000000 with 0 decimals of accuracy␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=0.0°C Humidity=2.0%␛[0m
```

I added DHT_MODEL_TYPE,  It works fine
```
  auto dht = App.make_dht_sensor("temp", "humi",D2);
  dht.dht->set_dht_model(sensor::DHT_MODEL_DHT22);    //new line added

```
```
�␓␒nn␄l`␂␎␂nr����␀␌␌b␌�␏l␏{��n␄␌c␄�␏l�#cb␌␌��␃s$`␂��o�␂␛[0;32m[I][log_component:055]: Log initialized␛[0m
␛[0;32m[I][application:051]: Running through setup()...␛[0m
␛[0;35m[C][status_led:026]: Setting up Status LED...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 16 with mode OUTPUT␛[0m
␛[0;35m[C][wifi:031]: Setting up WiFi...␛[0m
   ...
␛[0;35m[C][mqtt.client:024]: Setting up MQTT...␛[0m
    ...
␛[0;32m[I][mqtt.client:137]: MQTT Connected!␛[0m
␛[0;35m[C][sensor.dht:033]: Setting up DHT...␛[0m
␛[0;35m[C][esphal:067]:     GPIO Pin 4 with mode OUTPUT␛[0m
␛[0;35m[C][component:243]:     Update interval: 15000ms␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'humi'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '%'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 0␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: 'mdi:water-percent'␛[0m
␛[0;35m[C][sensor.mqtt:027]: Setting up MQTT Sensor 'temp'...␛[0m
␛[0;35m[C][sensor.mqtt:029]:     Expire After: 675s␛[0m
␛[0;35m[C][sensor.mqtt:031]:     Unit of Measurement: '°C'␛[0m
␛[0;35m[C][sensor.mqtt:032]:     Accuracy Decimals: 1␛[0m
␛[0;35m[C][sensor.mqtt:033]:     Icon: ''␛[0m
␛[0;32m[I][application:088]: You're running esphomelib v1.8.2␛[0m
␛[0;32m[I][application:099]: Running through first loop()␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=21.3°C Humidity=53.9%␛[0m
␛[0;36m[D][sensor.mqtt:094]: 'temp': Pushing out value 21.300001 with 1 decimals of accuracy␛[0m
␛[0;36m[D][sensor.mqtt:094]: 'humi': Pushing out value 53.900002 with 0 decimals of accuracy␛[0m
␛[0;32m[I][application:115]: First loop finished successfully!␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=21.2°C Humidity=53.5%␛[0m
␛[0;36m[D][sensor.dht:054]: Got Temperature=21.2°C Humidity=54.3%␛[0m
```

OttoWinter, 2018-11-27:
Closing as esphomelib now informs the user that they should add an explicit model type if reading the sensor fails.

closed 2018-11-27 by OttoWinter