esphome/esphome-core #149: A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header

philko123, 2018-09-20:
I try to flash my esp01 thru the hassio esphomelibyaml-addon

Platform: 
- Raspi3
- Hassos
- esp01 connected to ttyusb01

output
`
INFO [esphomeyaml.config] Reading configuration...
INFO [esphomeyaml.__main__] Generating C++ source...
INFO [esphomeyaml.__main__] Compiling app...
INFO [esphomeyaml.__main__] Running:  platformio run -d /config/esphomeyaml/meinschalter001
[Thu Sep 20 07:32:44 2018] Processing meinschalter001 (platform: espressif8266; board: esp01_1m; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
SYSTEM: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <Hash> v1.0
|-- <esphomelib> v1.7.0 #bd07644
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Hash> v1.0
|   |   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <FastLED> v3.1.8 #d416194
|   |   |-- <EspSoftwareSerial> v3.3.1
|   |-- <Wire> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoOTA> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> v1.0
Calculating size .pioenvs/meinschalter001/firmware.elf
text	   data	    bss	    dec	    hex	filename
324000	  13244	  30648	 367892	  59d14	.pioenvs/meinschalter001/firmware.elf
========================= [SUCCESS] Took 45.32 seconds =========================

********************************************************************************
There is a new version 3.6.0 of PlatformIO available.
Please upgrade it via `platformio upgrade` or `pip install -U platformio` command.
Changes: http://docs.platformio.org/en/latest/history.html
********************************************************************************

********************************************************************************
There are the new updates for platforms (espressif8266)
Please update them via `platformio platform update` command.

If you want to manually check for the new versions without updating, please use `platformio platform update --only-check` command.
********************************************************************************

INFO [esphomeyaml.__main__] Successfully compiled program.
INFO [esphomeyaml.__main__] Uploading binary...
INFO [esphomeyaml.__main__] Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /config/esphomeyaml/meinschalter001/.pioenvs/meinschalter001/firmware.bin
esptool.py v2.4.0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
`
What can I do?

OttoWinter, 2018-09-25:
Hmmm, I don't know. Can you confirm the ESP8266 works, maybe with another firmware? The two things I can think of are: the ESP8266 is dead or the ESP8266 isn't getting enough power over USB.

OttoWinter, 2018-11-27:
Closing this issue as it looks a lot like a hardware issue.

closed 2018-11-27 by OttoWinter

Nornode, 2019-06-20:
Hey, I came across this issue.
Thought It was my ESP8266, changed it, and have the same problem (Nothing before soldering my sensors). Just a thought @philko123 How long cable did you attach to your i2c sensor?
Mine is +200cm and thougth this might be the issue with problems for pull-ups and resistors et.al. to get started.... This might be the hardware issue we are facing?

OttoWinter, 2019-06-20:
@Nornode Please don't use old closed issues for further discussions - see the new issue tracker please.