esphome/esphome-core #153: Support RC522

thoscut, 2018-09-26:
It's a great 1.8.0 release and the support for NFC readers is a great addition.
I wondered if you plan to support the RC522 as well?

OttoWinter, 2018-09-27:
Just ordered one and will try to see how difficult it is when it arrives (shipment from china, could take some time)

thoscut, 2018-09-27:
Hey Otto, that’s great. I could donate you a spare one, shipping from Germany. 😉

rradar, 2018-11-16:
And here is a link for a working integration/project for the dirt cheap ($1.5 with shipping and tags!) RC522:
- https://github.com/esprfid/esp-rfid

OttoWinter, 2018-11-27:
I just had a go at it using [this library](https://github.com/pkourany/MFRC522_RFID_Library/tree/master/src) as a starting point. However, the RC522 seems to be quite a lot "weirder" than the PN532, and the PN532 was _already_ hard to implement since its protocol is quite hard to understand. And that's *before* having to go through all the debugging.

Especially [this method](https://github.com/pkourany/MFRC522_RFID_Library/blob/master/src/MFRC522.cpp#L467) here is a prime example of bad source. It has lots of comments, yes, but it's still very hard to understand and uses ancient style variable re-use.

Putting this on hold until I figure out a better way to implement it.