esphome/esphome-core #158: [Question] How to isolate binary sensor press, release, click and double click. #question

skeptizzle, 2018-09-27:
Hi. 
Is it possible to isolate on_press, on_release, on_click and on_double_click so i can publish eg. on_double_click to MQTT instead of getting all the messages?

My binary_sensor is setup like this:

```
  - platform: gpio
    name: "Wall Kitchen TL"
    on_press:
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: "PRESSED"
    on_release:
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: "RELEASED"
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: "CLICK"
    on_double_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: "DOUBLE CLICK"
    pin:
      number: D3
      mode: INPUT_PULLUP
      inverted: True
```

OttoWinter, 2018-09-27:
I don't really understand what you mean by isolate. Do you only want to get the, for example, `on_double_click` MQTT messages? Because then this would work:

```yaml
  - platform: gpio
    name: "Wall Kitchen TL"
    on_double_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: "DOUBLE CLICK"
    pin:
      number: D3
      mode: INPUT_PULLUP
      inverted: True
```

skeptizzle, 2018-09-27:
What I mean by isolate is that I want a different state published depending on the type of press on the button. I just can't figure out how when I'm reading the documentation.

Let me try differently, when I press and hold the button down, I want to publish, let's say, "pressed" to some/topic, but I don't want "pressed" if I "double click". 

I hope I'm making sense.. :-\

brandond, 2018-11-05:
I think what they're asking for is a generic on_event trigger that would send the last event name, rather than having to define handlers for a bunch of different events. Something like this:

```yaml
  - platform: gpio
    name: "Wall Kitchen TL"
    id: wall_kitchen
    on_event:
      then:
        - mqtt.publish:
            topic: wall_kitchen/button/wall_kitchen_tl/state
            payload: !lambda >-
              return id(wall_kitchen).last_event;
```

skeptizzle, 2018-11-05:
@brandond yes! Exactly that! I couldn't get the right wording. Thanks!

closed 2018-11-09 by OttoWinter