esphome/esphome-core #165: Pulse counter

voicevon, 2018-10-01:
I am using pulse , the value unit is "pulse/min". I hope there are two measure unit. First is "Total pulses count" ,it is for "How many pulses the sensor has caught". Second is "Rate/Ratio", Just same requirement as currently what the pulse sensor is doing. Further more, I want the third value like "Acceleration", to show how much it's going faster/slower.  
I tried to modify the code, But it should be one GPIO connected, Like the behavior of DHT-11, two sensors with one wire. This will be a challenge for me. @OttoWinter , Do you think my idea is doable?

BTW: one document error at https://esphomelib.com/api/sensor/pulse_counter.html  at "Example usage"
```
strom_warme.mqtt->set_unit_of_measurement("kW");
strom_warme.mqtt->clear_filters();
strom_warme.mqtt->add_multiply_filter(0.06f); // convert from Wh pulse to kW
```
the mqtt is wrong, should be pcnt.

OttoWinter, 2018-10-04:
Yes, it is possible:

```yaml
sensor:
  - platform: pulse_counter
    filters:
      - lambda: |-
          static float total_value = 0.0;
          // 0.25 because of update_interval 15s
          total_value += x * 0.25;
          return total_value;
```

The acceleration would be more or less the same, but I'll leave it as an exercise to the reader :P

voicevon, 2018-10-06:
Yes, Lambda can **convert** from "pulse/min" to "Total pulse count". Saying, I can only get one of them,  I got this, I lost that.   But my request is： The sensor provides two or three values that they represent different meaning at the same time.

voicevon, 2018-10-14:
I am not able to use Yaml right now. I read a lot source code of `TemplateSensor`, but still can't archive . The main reason is I don't know how to write an instance of 
`void set_template(std::function<optional<float>()> &&f)`  
Could you give a two/three lines example? I guess the code should be  like below. Thanks.

```
   auto water= App.make_pulse_counter_sensor("flow", GPIOInputPin( 18,INPUT_PULLUP));  
        water.pcnt->clear_filters();  
 auto rate = App.make_template_sensor("flow/hour");
 second.template_->set_template(&& water.pcnt->get_value());  //My target line, right?
```

voicevon, 2018-10-22:
I got it .
```
  Application::MakeTemplateSensor target = App.make_template_sensor("target",2000);
      target.template_->set_template([=]() -> optional<float> {
      static float result = 0.0;
      result = source->value + 123;   //result = anyfunction( source->value);
      return result; 
  });
```

closed 2018-10-24 by voicevon

MrUkleja, 2019-04-22:
Sorry to re-open this
I am trying to achieve similar - get "Total pulses couter"

I tried what @OttoWinter  suggested above. With one change id(pulse).value -> id(pulse).state (as value was depreciated) 
But I'm getting 'unknown value' in Hass.io all the time 
Could you help me solve this?

Here is my code:

```
sensor:
  - platform: pulse_counter
    pin: D5
    unit_of_measurement: 'W'
    name: 'Pobór prądu'
    id: pulse
    filters:
      - multiply: 9.375
  - platform: template
    name: "Zużycie prądu całkowite"
    unit_of_measurement: 'kWh'
    lambda: |-
      static float total_value = 0.0;
      total_value += id(pulse).state;
      return total_value;
```