esphome/esphome-core #168: Weird button activation at start

TheJulianJES, 2018-10-01:
When there's a button hooked up with a 10k pulldown resistor (to ground) and having 3.3v to the data pin D2 on the NodeMCUv2 (esp8266) when the button is switched on, it turns the light on//off as expected, but for some reason at boot (not reboot, maybe state saves there?) the buttons seems to be pressed (thus turning the light). Although this can be solved by using a filter, like I did below.

Not sure if this is an issue, but I expect that other people will also experience this.

(Note: I'm pretty sure the on_press function was not affected when I tried)

(And I also could have used an internal pullup, but the build already existed and it's hard to change it now.)

Relevant esphomeyaml config entry:
```
light:
  - platform: fastled_clockless
    name: "led_9_strip"
    id: leds
    chipset: WS2811
    pin: D3
    num_leds: 12
    rgb_order: GRB
    gamma_correct: 1.0
    effects:
      - random:
      - strobe:
      - flicker:
      - fastled_rainbow:
      - fastled_color_wipe:
      - fastled_scan:
      - fastled_twinkle:
      - fastled_random_twinkle:
      - fastled_fireworks:
      - fastled_flicker:

binary_sensor:
  - platform: gpio
    pin:
      number: D2
    id: toggle_button_one
// When there's no filter present, the buttons seems to be "pressed" at every start, thus turning the light on at start
    filters:
      - lambda: >-
          if (millis() > 500) {
            return x;
          } else {
            return true;
          }
    on_click:
    - min_length: 0ms
      max_length: 349ms
// Called at boot, when the filter is not present
      then:
        - light.toggle:
            id: leds
    - min_length: 350ms
      max_length: 1000ms
      then:
        - light.turn_on:
            id: leds
            brightness: !lambda >-
                static bool state;
                
                state = !state;
                if (state) {
                  return 1.0;
                } else {
                  return 0.5;
                }
```

OttoWinter, 2018-10-04:
I can't replicate this behavior. For me, the pin is low on boot-up as expected :(

TheJulianJES, 2018-10-04:
Mhm. That's weird. I just rebuilt it (https://www.arduino.cc/en/Tutorial/Button was what I used in the original build and rebuilt now) and it still happens for some reason.
(On a NodeMCU though, esp8266).
It only happens when using on_click (like in my posted esphomeyaml config), but not when using on_press.
The button is connected on one side to D2 and also to a 10k resistor between ground.
The other side is connected to 3.3V. (Like in the picture on the arduino website).

OttoWinter, 2018-11-27:
Closing this issue as it looks a lot like a hardware issue to me. Please reopen if you can give me a config/setup with which this issue can be reproduced

closed 2018-11-27 by OttoWinter

TheJulianJES, 2019-01-19:
Seems to be fixed by #425