esphome/esphome-core #170: There is no posibility to connect to wifi with WIFI 802.1x EAP (WPA2-Enterprise)

komw, 2018-10-02:
I would be great, if esphomelib devices can use a WPA2-Enterprise Wifi passwords. Currently only simple WPA2 Personal is supported.
I notived that there is a support in arduino for wpa2-enterprise for ESP8266 (https://github.com/esp8266/Arduino/issues/1032), so it shouldn't be hard to add it.

komw, 2018-10-02:
But for the other hand: https://github.com/opendata-stuttgart/meta/issues/84 they said that is could be immpossible due to CPU limits

OttoWinter, 2018-10-04:
A PR that fixes this would be welcome :P

I can't really do this because I don't have a WPA-2 enterprise network to test it on and I'm pretty sure that espressif's implementation is quite buggy/doesn't work well. Also, it would need to also work on the ESP32 which could end up being a big mess.

brandond, 2018-12-29:
No comment on how well this would work with ESP32, but it is supported in the ESP8266 core, see:
https://github.com/espressif/ESP8266_NONOS_SDK/blob/release/v3.0.0/examples/wpa2_enterprise/user/user_main.c