esphome/esphome-core #171: Pulse counter GPIO as pull-up input

voicevon, 2018-10-02:
In my case. The pulse counter is working very well. But I have to connect a pull up resistor.  I tried to find out where to setup the pin enable the internal pull-up. But I can't find it. Please help.

TheJulianJES, 2018-10-02:
I'm not sure if you're also using esphomeyaml to generate esphomelib code, so you could look at this article in the docs: https://esphomelib.com/esphomeyaml/guides/configuration-types.html#config-pin-schema
It describes what you would add to the "mode" option in the pulse counter.

voicevon, 2018-10-03:
Thanks for the link.  I am not using esphomeYaml, I want to modify the source code at application code level.(Means I don't want to modify the code in /Lib and sub folder). For example, I expect I can do the configuration like:

  ```
  auto water = App.make_pulse_counter_sensor("water2",D8);
    water.pcnt->set_pin_mode(INPUT_PULLUP);   //There is no set_pin_mode();
```

TheJulianJES, 2018-10-03:
Haha, wasn't a stupid question. Glad you found it. ^^

OttoWinter, 2018-10-03:
@voicevon I don't think that'll work (the pulse counter will override the pin mode). Almost all pin options in the esphomelib API take a [`GPIOPin`](https://esphomelib.com/api/core/helpers.html#_CPPv3N12GPIOInputPin12GPIOInputPinE7uint8_t7uint8_tb) instance. 

```yaml
App.make_pulse_counter_sensor("water2", GPIOInputPin(D8, INPUT_PULLUP));
```

closed 2018-10-08 by voicevon