esphome/esphome-core #173: DSC Keybus library support

mzac, 2018-10-05:
It would be very interesting to integrate this DSC Keybus library into esphomelib/esphomeyaml:

https://github.com/taligentx/dscKeybusInterface

Not sure how difficult it would be but then it could be a lot easier to configure an ESP8266 to talk to a DSC alarm panel!

OttoWinter, 2018-10-08:
Puh that's a big library. I also don't have the hardware to test it (nor will I buy it - seems expensive 🙃). So I'll put it into the low-priority stuff. Sorry, but I'd need to read into how that DSC keybus protocol works and that will be a lot of work.

mzac, 2018-10-16:
Yes, it is big and maybe more complicated, however I haven't had a full chance to look how esphomelib works but the examples provided by the author for the arduino side are quite simple.  I only had to make a few changes to the code and upload it to a D1 mini and it worked right away.  For now I'll keep it like this but it would be a nice addition to esphomelib.  The alternatives out there are a commercial product called 'Eyeezon' http://www.eyezon.com/ but this is a $120+ solution where the D1, a perfboard, some resistors, wires and a transistor make it all under $10 !

Let me know if there is anything you'd like me to test as I do have one of these panels.

Thanks!