esphome/esphome-core #190: Lights turn off under ~24 brightness

DarkFox, 2018-10-10:
I'm guessing due to the way the gamma correction is implemented, setting the light brightness too low causes them to turn off completely. Changing the gamma allows you to go further down, but at the cost of less correction.

OttoWinter, 2018-10-10:
Yes... but this is a hardware limitation. I guess you're using the esp8266 pwm platform, right?

Well that can only output values from `0` to `1023`, and with a gamma correction factor of `2.8` (default), you get an output value of `(x/255)^2.8 * 1023` which for `x = 24` equals `1.369` or rounded `1`.

So it's impossible to have lights below that brightness for the ESP8266 PWM platform. For alternatives, please try the ESP32 `ledc` (much higher range) or pca9685.

DarkFox, 2018-10-11:
Unfortunately, I'm using h801 LED controllers, and already have a significant amount of them, so changing hardware isn't an option. I'll just have to live with it I guess.

brubaked, 2018-10-31:
I ran into the same issue. I was wondering if it is possible to rescale the pwm range from 0 - 1023 to say 25 - 1023.

OttoWinter, 2018-11-13:
I just created https://github.com/OttoWinter/esphomelib/commit/e38e6bf788b96762c789c85ed9a97318ea600dfc which should solve the problem: Now you can adjust the PWM frequency, so if you want more colors to be representable, then just lower the PWM frequency.

closed 2018-11-13 by OttoWinter