esphome/esphome-core #201: pn532 NewTag echo not reliable

wuub, 2018-10-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** just the board, 

**ESP (ESP32/ESP8266/Board/Sonoff):** tested on esp8266 d1_mini and esp32 lolin32

**Affected component:** pn532

**Description of problem:** Fresh pn532 board + 21 mifare cards. I tried to scan all of them to get their IDs, but encountered a problem. Some of the cards would apparently not scan. I bumped the logger verbosity to   level: VERBOSE and noticed that they do all **do scan correctly [sic!]**. 

Each and every one of them generates `[V][pn532:083]: ATQA: 0x0004, SAK: 0x08` line, but only some (and repeatably so) generate `[21:34:22][D][pn532:117]: Found new tag 'A6-DF-51-DE'`. 
Out of a stack of 21 cards, 1/3 or 1/4th reliably generate `new tag` message while the rest is silent. At least **until I restart** the board, after a restart a set of cards generating `new tag` output changes randomly (but stays consistent for the whole uptime of the board)

After a few restarts I managed to scan all the cards, after I created a binary_sensors for all of them **ALL WORK RELIABLY** generating mqtt messages each and every time.

**Additional information:** if I had to guess, this line looks suspicious https://github.com/OttoWinter/esphomelib/blob/2ffe6c1b23fb98012e95d39d98db8e04d783a771/src/esphomelib/binary_sensor/pn532_component.cpp#L96

OttoWinter, 2018-10-12:
OMG! Thank you!

I was looking at the PN532 code a long time because of https://github.com/OttoWinter/esphomeyaml/issues/147, but now it all makes sense! I was looking at the PN532 protocol layer the entire time because I couldn't reproduce it with my 5 tags I have, but the problem was in an entirely different place. 

Yes, that line is incorrect, what a single character can do 😮

wuub, 2018-10-12:
Glad I could help :) I love tracking down this kind of bugs, I was hoping to use my oscilloscope, but your logging is too good :) 

I f I can take one more minute of your time, is there a way to push all the pn532 binary sensors to a single topic but with a tag ID as payload? will `on_press` work?

OttoWinter, 2018-10-12:
Uhm, I think that's a bit more difficult, but this should work:

```yaml
binary_sensor:
- platform: pn532
  id: tag1
  uid: 74-10-37-94
  on_press:
  - mqtt.publish:
      topic: the/topic
      payload: '74-10-37-94'
```

Basically, this will mark the binary sensor as `internal` (since no `name` is specified) so that it doesn't create the normal MQTT messages and each time it's pressed it will send the specified payload out.

It's a bit long, but will do the job.

wuub, 2018-10-12:
Fantastic. Thank you, this is all I needed.

closed 2018-10-13 by OttoWinter