esphome/esphome-core #216: Moving average window filters push the first value without averaging

r-jordan, 2018-10-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
I have several WemosD1 and Nodemcu compiled in a Hass.io Addon docker.

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266

**Affected component:**
https://esphomelib.com/esphomeyaml/components/sensor/index.html#sensor-filters
https://esphomelib.com/api/sensor/index.html#classsensor_1_1_exponential_moving_average_filter
https://esphomelib.com/api/sensor/index.html#classsensor_1_1_sliding_window_moving_average_filter

**Description of problem:**

It occurs with the default sensor filter or Sliding Window filter and the Exponential filter. When it receives the first value from the sensor it's pushed without being averaged with any other value. I think the expected would be to wait until the send_every parameter number of values is reached.

The function of the send_every parameter is not altered because values will be pushed at the same periodicity but the first value pushed will be delayed until at least it can be averaged with some other received sensor values in the moving average filter.

**Problem-relevant YAML-configuration entries:**

```yaml
  - sliding_window_moving_average:
      window_size: 15
      send_every: 15
  - exponential_moving_average:
      alpha: 0.1
      send_every: 15
```

**Traceback (if applicable):**

```

```

**Additional information:**
It can be fixed with minimal modification in the file esphomelib/src/esphomelib/sensor/filter.cpp
replacing `send_at_(send_every - 1)` by `send_at_(0)` at the constructors of the ExponentialMovingAverageFilter and SlidingWindowMovingAverageFilter classes.

OttoWinter, 2018-10-14:
Yes that "waiting" behavior was also the behavior in the first couple of releases of esphomelib.

But the problem with that behavior is for new users. When you create your first sensor, the sliding window moving average filter will be applied by default. When you then first upload that firmware, many users were wondering why their sensor values weren't showing up in home assistant until after a few minutes or so. So I decided to make the first value "special" and push it out immediately.

I don't really want to change the *default* behavior as per the reason above. But ... I think just adding an option like `send_first_immediately` would provide a configuration option for users who do need the filter to truly average the values.

r-jordan, 2018-10-14:
Ah, ok, I understand.
In my case I'm making a solar powered device that has to deep sleep and I don't want it to push any value every time it just wakes up. A new parameter like `send_first_immediately` or like `send_first_at` (more flexible with the minimal number of sensor values needed to read) would be very useful.
Thanks for your great work Otto!

r-jordan, 2018-10-20:
Hi @OttoWinter , I would like very much to have it working in the next release.
May be I can help you creating a pull request to implement this optional parameter in the moving average filters.

closed 2018-10-26 by OttoWinter