esphome/esphome-core #235: Add amplifier setting for PN532

thoscut, 2018-10-21:
The reading distance of the PN532 can be too short.

**Describe the solution you'd like**
Add the amplifier/sensitivity setting to allow using increased distance reading as outlined in the data sheet.

OttoWinter, 2018-10-23:
Interesting. I didn't know the PN532 had such an option.

Have you used this register before? Because I don't want to sift through the hundreds of pages of the datasheet to find out how to write to a value to a register :P It's probably quite simple, but if you've done it before that would be awesome :D

thoscut, 2018-10-24:
Unfortunately not, there seems also to be no other publicly available implementation. One of the few sources found referencing the register is https://github.com/gefangenimnetz/mopidy-nfcread/blob/master/mopidy_nfcread/nfc/clf/pn53x.py but it's not used in the implementation.

The feature is at least integrated in the latest (hardware?) revision (Rev. 3.6 — 28 November 2017) of the PN532 (https://www.nxp.com/docs/en/nxp/data-sheets/PN532_C1.pdf), see 8.6.8/8.6.23.36 and should allow to configure the receiver gain, which, if I didn't get it wrong, should also increase of the reading distance. 
The older datasheets only contain the *RF Level detector* in the feature list but without the information about the register used for configuration. Not sure if the are differences in the hardware/firmware and there's a need for detecting the revision of the chipset/firmware.