esphome/esphome-core #237: TemplateSensor returns string/text

voicevon, 2018-10-24:
Currently, the templateSensor is only return datatype with float.
Here is an use case why I want it returns a string: I created a  UpTimerSensor, and I monitored it on my mobile phone with an app named "mqtt dashboard". They worked very well with each other. But when I see it shows 3605 seconds, I hope it is "1h0m5s". 
I know one solution is to modify the display text in MQTT dashboard, But I have to do it many times if I have multi phones. So, I created a template sensor, with lambda as below
```
 Application::MakeTemplateSensor myUptimer = App.make_template_sensor("MyUpTimer",2000);
     myUptimer.template_->set_template([=]() -> optional<std::string> {
       int hour = sourceUpTimer.uptime->value / 3600;
       int minute = sourceUpTimer.uptime->value /  60;
       int second = (int) (sourceUpTimer.uptime->value) % 60;

      std::ostringstream aaa;
       std::string result = std::to_string(hour) +"h" + std::to_string(minute) +"m"+std::to_string(second) +"s";
     return result; 
 });
```

But this doesn't work, because templateSensor is always return float. I don't know how much codes is involved to this return datatype, Is it possible/hard/easy to get this feature? Or is there any alternative solution for my case liked?

OttoWinter, 2018-10-26:
https://beta.esphomelib.com/esphomeyaml/index.html#text-sensor-components ;-)

voicevon, 2018-10-31:
Thank you so much  👍

closed 2018-10-31 by voicevon