esphome/esphome-core #244: MQTT+ap fallback mode do not work together

wvolz, 2018-10-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip installed version of esphomeyaml

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266 NodeMCU v2

**Affected component:**
https://esphomelib.com/esphomeyaml/components/wifi.html (and maybe mqtt)

**Description of problem:**
When MQTT and AP fall back mode are enabled my ESP8266 ends up in a reboot loop with MQTT not connecting.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
    ssid: 'SSID'
    password: 'password'
    ap:
        ssid: 'ESP-recovery'

  mqtt:
    broker: 'broker.somedomain.com'
    username: ''
    password: ''
```

**Traceback (if applicable):**

```
[I][log_component:055]: Log initialized.
[C][ota:186]: There have been 4 suspected unsuccessful boot attempts..
[I][application:051]: Running through setup()....
[C][status_led:026]: Setting up Status LED....
[C][esphal:067]:     GPIO Pin 16 with mode OUTPUT.
[C][light.fast_led:039]: Setting up FastLED light....
[C][light.fast_led:046]:     Max refresh rate: 400.
[C][light.state:188]: Setting up light 'FastLED WS2811 Light'....
[C][wifi:031]: Setting up WiFi....
[I][wifi:276]: WiFi Connecting to 'SSID'....
[C][wifi:293]:     STA Hostname: 'hostname'.
[C][ota:027]: Setting up OTA....
[C][ota:028]:     port: 8266.
[C][ota:033]:     hostname: 'hostname'.
[W][ota:124]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts.
[C][mqtt.client:024]: Setting up MQTT....
[C][mqtt.client:025]:     Server Address: broker.somedomain.com:1883.
[C][mqtt.client:026]:     Username: ''.
[C][mqtt.client:027]:     Password: ''.
[C][mqtt.client:030]:     Client ID: 'hostname-1dde5f1cc4a1'.
[C][mqtt.client:032]:     Discovery prefix: 'homeassistant'.
[C][mqtt.client:033]:     Discovery retain: true.
[I][wifi:324]: WiFi connected!.
[C][wifi:326]:     SSID: SSID.
[C][wifi:328]:     BSSID: xx:xx:xx:xx:xx:xx.
[C][wifi:329]:     Channel: 6.
[C][wifi:330]:     IP Address: 192.168.50.149.
[C][wifi:331]:     Subnet: 255.255.255.0.
[C][wifi:332]:     Gateway: 192.168.50.1.
[C][wifi:333]:     DNS1: 192.168.50.1.
[C][wifi:334]:     DNS2: 0.0.0.0.
[D][wifi:338]: Disabling AP....
```

**Additional information:**
The above loop continues until the device goes into safe mode.

OttoWinter, 2018-11-08:
Well there are possible scenarios when the MQTT server is **on the AP** created by the ESP. For example I use that for debugging when I don't have a router around: I let the ESP create the AP and connect to it using my laptop which also runs mosquitto.

OttoWinter, 2019-01-08:
Fixed

closed 2019-01-08 by OttoWinter