esphome/esphome-core #245: WiFi IP Address sensor

jquatier, 2018-10-29:
I was looking at the beta docs and it looks like since strings will be supported with the Text Sensor, it would be great to add a sensor to report the current IP address (string), similar to the WiFi Signal Strength sensor. This could be used on OLED/LCD displays, etc. At a high level I think I just need to call WiFi.localIP(), but let me know if this is a good candidate for a sensor. Thanks!

OttoWinter, 2018-11-08:
I don't think I'll add this as a component as it's super easy to do with a template:

```yaml
text_sensor:
- platform: template
  # ...
  lambda: 'return WiFi.localIP().toString().c_str();'
```

closed 2018-11-08 by OttoWinter

jquatier, 2018-11-08:
Perfect. Thanks!