esphome/esphome-core #247: Document how to setup wifi ap mode with and without station mode

JAAlperin, 2018-11-02:
The code below does not provide an access point.  I assume simultaneous station and access point modes are established and the access point is turned off.  How can I get both mode active together?  If not that, how can I get only AP mode?  Sorry, I cannot find that documented.

`esphomeyaml:
  name: nodemcu08
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version:
    repository: https://github.com/OttoWinter/esphomelib.git
#   branch: master
    tag: v1.8.2
  arduino_version: latest

logger: {}

mqtt:
 broker: 192.168.0.100
 username: nodemcu08
 password: ''
 port: 1883

ota:
  password: XXXXXXXX
  safe_mode: true

wifi:
  ssid: ddwrt2G
  password: 'XXXXXXXX'
  domain: .alperin.lan
  manual_ip:
    static_ip: 192.168.0.143
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  ap:
    ssid: ddwrt2Gap
    password: 'test1234'
`

OttoWinter, 2018-11-02:
That's not possible right now. Please state your use-case

JAAlperin, 2018-11-02:
Thank you, Otto.

In my garage I have a nodemcu using your excellent esphomeyaml system to 
monitor temperature and humidity for HASS. I would also like to have an 
access point in my garage so that my car can connect to my home network 
without opening the garage door. I could accomplish that with a spare 
router or an rPi, but it seems natural to just add the AP to the nodemcu 
already perfectly placed. The esphomelib api docs seems to suggest this 
is possible.

JAAlperin, 2018-11-02:
> Setting up ESPs as repeaters is very difficult with ESPs and not a
> goal of esphomelib. You can use
> https://github.com/martin-ger/esp_wifi_repeater if you want, but the
> data throughput (and stability) of that is very bad.

Thanks, again, Otto.

I'll just go with a Raspberry Pi.  I need a connection, not really 
throughput.

Regards,

Jeff Alperin

H: 561-666-7789
C: 561-797-9876

"It's not rocket surgery!" - Oscar Leroy
"You know my rule about circus music!" - Lacey Burroughs