esphome/esphome-core #248: Suport for HM-10

Esiravegna, 2018-11-05:
As the BLE is not still fully available until the espressif core got updated, and thinking on the ESP8266 chipset, do you think it is possible to adapt the already existing BLE controller to use [this](https://github.com/dinosd/BLE_PROXIMITY/blob/master/examples/proximity/proximity.ino) library for the HM-10 module?

Thanks a lot for your hard work!

OttoWinter, 2018-11-08:
Duplicate of https://github.com/OttoWinter/esphomeyaml/issues/77 - and what do you mean by "As the BLE is not still fully available until the espressif core got update" ?

closed 2018-11-08 by OttoWinter

Esiravegna, 2018-11-08:
For using BLE I'll have to use the espressiff old core, 1.0something. When using that, I got a lot of crashes and reboots, however, I don't know if these are caused by the BLE or the espressif core.
Thanks!