esphome/esphome-core #250: HLW8012 / CSE7759: glitches in measurement: negative values

RolluS, 2018-11-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Unraid/Docker installation on i686 home server, with separate containers for Home assistant, MQQT, and ESPhomeyaml, all from official repos.

**ESP (ESP32/ESP8266/Board/Sonoff):**

Sonoff POW R1 v2 (CSE 7759 metering chip instead of HLW8012, shall be supported by Xoze Perez's library) 

**Affected component:**

https://esphomelib.com/esphomeyaml/components/sensor/hlw8012.html

**Description of problem:**
Hello Otto, thank you for your hard work, ESPHomeyaml and ESPHomelib are now the only control system in my home automation (using weemos or sonoff for heating, some lightning..). Lighting is done by yeelight and sensor are through xiaomi gateway at the moment.

Well, I unfortunately have an issue with sonoff pow: 
Regularly, way out of window data is retrieved. It may be glitches from the measuring device itself as per my readings.
I've tried with differents update_intervals.

Positive values seems accurate.
Is it possible to discard negative values? Using a lambda filter?
Maybe a condition could be added in the code.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: hlw8012
    sel_pin: 5
    cf_pin: 14
    cf1_pin: 13
    #commented to clear logs
    # current:
    #   name: "Courant Lave-linge"
    #   filters:
    voltage:
      name: "Tension Lave-linge"
      filters:
        - multiply: 1.1233
    # power:
    #   name: "Puissance Lave-linge"
    #   filters:
    update_interval: 5s
    voltage_divider: 3000 #has no effect (original 2351)
    current_resistor: 0.001
```

**Traceback (if applicable):**

```
[15:58:48][C][light.mqtt_json:020]: Setting up MQTT light...
[15:58:48][D][mqtt.client:188]: Subscribing to topic='lave_linge/light/connexion_lave-linge/command' qos=0 with JSON...
[15:58:51][D][sensor.hlw8012:058]: Got power=2.1W, voltage=161.7V
[15:58:51][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 181.680649 with 1 decimals of accuracy
[15:58:56][D][sensor.hlw8012:058]: Got power=0.0W, voltage=205.7V
[15:58:56][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.071350 with 1 decimals of accuracy
[15:59:02][D][sensor.hlw8012:058]: Got power=0.0W, voltage=206.0V
[15:59:02][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.346771 with 1 decimals of accuracy
[15:59:07][D][sensor.hlw8012:058]: Got power=2.1W, voltage=206.0V
[15:59:07][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.438583 with 1 decimals of accuracy
[15:59:11][D][sensor.hlw8012:058]: Got power=0.0W, voltage=205.5V
[15:59:12][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 230.887741 with 1 decimals of accuracy
[15:59:16][D][sensor.hlw8012:058]: Got power=0.0W, voltage=206.4V
[15:59:17][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.805786 with 1 decimals of accuracy
[15:59:22][D][sensor.hlw8012:058]: Got power=2.1W, voltage=206.4V
[15:59:22][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.897598 with 1 decimals of accuracy
[15:59:26][D][sensor.hlw8012:058]: Got power=0.0W, voltage=206.3V
[15:59:27][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.713974 with 1 decimals of accuracy
[15:59:36][D][sensor.hlw8012:050]: Got power=0.0W, current=0.1A
[15:59:41][D][sensor.hlw8012:050]: Got power=0.0W, current=0.1A
[15:59:46][D][sensor.hlw8012:050]: Got power=2.1W, current=0.1A
[15:59:51][D][sensor.hlw8012:050]: Got power=2.1W, current=0.1A
[15:59:57][D][sensor.hlw8012:050]: Got power=0.0W, current=0.1A
[16:00:01][D][sensor.hlw8012:050]: Got power=0.0W, current=0.1A
[16:00:07][D][sensor.hlw8012:050]: Got power=2.1W, current=0.1A
[16:00:12][D][sensor.hlw8012:050]: Got power=0.0W, current=0.1A
[16:00:22][D][sensor.hlw8012:058]: Got power=0.0W, voltage=205.7V
[16:00:22][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 231.071350 with 1 decimals of accuracy
[16:00:27][D][sensor.hlw8012:058]: Got power=0.0W, voltage=206.6V
[16:00:27][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 232.081223 with 1 decimals of accuracy
[16:00:31][D][sensor.hlw8012:058]: Got power=2.1W, voltage=206.7V
[16:00:32][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 232.173004 with 1 decimals of accuracy
[16:00:36][D][sensor.hlw8012:058]: Got power=0.0W, voltage=207.5V
[16:00:37][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 233.091049 with 1 decimals of accuracy
[16:00:41][I][ota:157]: Boot seems successful, resetting boot loop counter.
**[16:00:42][D][sensor.hlw8012:058]: Got power=2.1W, voltage=-5148.3V
[16:00:42][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value -5783.118164 with 1 decimals of accuracy**
[16:00:46][D][sensor.hlw8012:058]: Got power=0.0W, voltage=207.0V
[16:00:47][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 232.540207 with 1 decimals of accuracy
[16:00:52][D][sensor.hlw8012:058]: Got power=2.1W, voltage=206.8V
[16:00:52][D][sensor.mqtt:080]: 'Tension Lave-linge': Pushing out value 232.264801 with 1 decimals of accuracy
```

**Additional information:**
Bad voltage measurement happens every few cycles in voltage mode.
Current measurement have same behavior but but occur few times a day.
`change_mode_every` parameter have no effect on this behavior

OttoWinter, 2018-11-08:
Fixed the day before in https://github.com/OttoWinter/esphomelib/commit/e666a592e3045de80ca5a5b23927a493bb32d01a :P

closed 2018-11-08 by OttoWinter

RolluS, 2018-11-11:
Haha, yes, this working perfectly since a couple of days now I switched to dev branch.

Thanks @OttoWinter