esphome/esphome-core #259: ESP32 BLE causes Exception  when using the v1.1.10 dev branch

Esiravegna, 2018-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 pip, windows 10

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP32s by DoIT

**Affected component:**
 BLE tracker

**Description of problem:**

Got these exception when enabling the BLE tracker component:
```
[14:15:15][W][mqtt.client:067]: MQTT Disconnected: TCP disconnected.
[14:15:15]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:15:15]Core 1 register dump:
[14:15:15]PC      : 0x4017123f  PS      : 0x00060d30  A0      : 0x801cc654  A1      : 0x3ffdefe0
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d '.\tower' -t idedata
[14:15:19]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000031  A5      : 0x0000d2fa
[14:15:19]A6      : 0x00000000  A7      : 0x0000059c  A8      : 0x8017b0d8  A9      : 0x3ffdefb0
[14:15:19]A10     : 0x3ffde360  A11     : 0x3ffc11c0  A12     : 0x3ffde3c0  A13     : 0x00000000
[14:15:19]A14     : 0x3ffde43c  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
[14:15:19]EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[14:15:19]
[14:15:19]Backtrace: 0x4017123f:0x3ffdefe0 0x401cc651:0x3ffdf000 0x4016b2f9:0x3ffdf020
WARNING Found stack trace! Trying to decode it
[14:15:19]
[14:15:19]Rebooting...
[14:15:19]ets Jun  8 2016 00:22:57
[14:15:19]
[14:15:19]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:15:19]configsip: 0, SPIWP:0xee
[14:15:19]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:15:19]mode:DIO, clock div:2
[14:15:19]load:0x3fff0018,len:4
[14:15:19]load:0x3fff001c,len:952
[14:15:19]load:0x40078000,len:6084
[14:15:19]load:0x40080000,len:7944
[14:15:19]entry 0x40080310
```

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_tracker:
  scan_interval: 30s
```

**Traceback (if applicable):**

```
[14:15:22]abort() was called at PC 0x4019c86f on core 1
[14:15:22]
[14:15:23]Backtrace: 0x40095ba8:0x3ffda860 0x40095dab:0x3ffda880 0x4019c86f:0x3ffda8a0 0x4019c8b6:0x3ffda8c0 0x4019b7eb:0x3ffda8e0 0x4019b8ee:0x3ffda900 0x4019be7d:0x3ffda920 0x4019bf9a:0x3ffda940 0x4019c0c5:0x3ffda980 0x400d6d89:0x3ffda9a0 0x400d6dcb:0x3ffda9c0 0x400d71ea:0x3ffda9e0 0x400d727a:0x3ffdaa20 0x400d72d6:0x3ffdaa40 0x400d8491:0x3ffdaa70 0x400d8a1b:0x3ffdaab0 0x400d8a79:0x3ffdaad0 0x400d8afd:0x3ffdab20 0x400d3a5d:0x3ffdab40 0x400d22cf:0x3ffdab90 0x401a166e:0x3ffdacb0
WARNING Found stack trace! Trying to decode it
[14:15:23]
[14:15:23]Rebooting...
[14:15:23]ets Jun  8 2016 00:22:57
[14:15:23]
[14:15:23]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:15:23]configsip: 0, SPIWP:0xee
[14:15:23]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:15:23]mode:DIO, clock div:2
[14:15:23]load:0x3fff0018,len:4
[14:15:23]load:0x3fff001c,len:952
[14:15:23]load:0x40078000,len:6084
[14:15:23]load:0x40080000,len:7944
[14:15:23]entry 0x40080310

```

**Additional information:**

Esiravegna, 2018-11-08:
Looks like a memory issue?
```
[14:41:37][W][wifi:377]: Restarting WiFi adapter...
[14:41:37][I][wifi:274]: WiFi Connecting to 'DOMUS-OCCULT'...
[14:41:37]E (12899) phy_init: failed to allocate memory for RF calibration data
[14:41:37]
[14:41:37]abort() was called at PC 0x40152b1d on core 0
[14:41:37]
[14:41:37]Backtrace: 0x40095ba8:0x3ffe12a0 0x40095dab:0x3ffe12c0 0x40152b1d:0x3ffe12e0 0x4018b2f6:0x3ffe1310 0x4018b390:0x3ffe1340 0x4018b67a:0x3ffe1370 0x4018b7f3:0x3ffe13a0 0x4008e024:0x3ffe13c0
WARNING Found stack trace! Trying to decode it
[14:41:37]
[14:41:37]Rebooting...
[14:41:37]ets Jun  8 2016 00:22:57
[14:41:37]
[14:41:37]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:41:37]configsip: 0, SPIWP:0xee
[14:41:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:41:37]mode:DIO, clock div:2
[14:41:37]load:0x3fff0018,len:4
[14:41:37]load:0x3fff001c,len:952
[14:41:37]load:0x40078000,len:6084
[14:41:37]load:0x40080000,len:7944
```

Esiravegna, 2018-11-08:
Disabling the BLE solved the issue, zero crashes.

Esiravegna, 2018-11-08:
Perhaps this is the cause? https://github.com/espressif/arduino-esp32/issues/1711

Esiravegna, 2018-11-08:
Tried to expand the things a bit, used a spare ESP32 with JUST a pir and the BLE, with this yaml:
```yaml
esphomeyaml:
  name: tv_rack
  board: nodemcu-32s
  platform: ESP32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk
  domain: '.local'

#esp32_ble_tracker:

mqtt:
  broker: 'mqtt.server.local'
  username: ''
  password: ''
  topic_prefix: 'tv_rack'

# Enable logging
logger:
  level: debug
ota:
  password: 'k1tch3n'
binary_sensor:
  - platform: status
    name: "tv rack status"
  - platform: gpio
    name: "tv rack PIR"
    pin: 12

```

Whit this configuration I don't have crashes anymore, compared with the original one that contains two custom sensors over i2c (max44009, and BMP180, and yes, I still can't make the built-in BMP85 component to report a correct result) and several pirs.  However, when un-commenting the ble, what I got are constant wifi crashes and re-connects.

OttoWinter, 2018-11-09:
> Perhaps this is the cause? espressif/arduino-esp32#1711

No, that's not the issue. esphomelib isn't repeatedly calling those methods.

However, I just noticed that we could do a small optimization for the memory usage of BLE (https://github.com/OttoWinter/esphomelib/pull/261). I personally haven't seen this problem on my own ESPs, what exact ESP32s are you using? Or more specifically: how much memory (RAM) do they have?

Esiravegna, 2018-11-09:
Thanks for your time! A [NodeMCU ESP32s](https://docs.zerynth.com/latest/official/board.zerynth.nodemcu_esp32/docs/index.html)
I believe that it got 520kb, as per the specs?

Esiravegna, 2018-11-09:
I can confirm that if I flash the very same board with just a PIR and the BLE enabled, it works flawlessly, with some wifi drops that very likely be caused by my infrastructure, so it is very likely a memory issue.

closed 2018-11-09 by OttoWinter

Esiravegna, 2018-11-10:
I'm afraid that the problem remains even after using the latest changes:
```
pressif"}}' retain=1)
[11:54:48]abort() was called at PC 0x4019cb77 on core 0
[11:54:48]
[11:54:48]Backtrace: 0x40095ba8:0x3ffee650 0x40095dab:0x3ffee670 0x4019cb77:0x3ffee690 0x4019cbbe:0x3ffee6b0 0x4019baab:0x3ffee6d0 0x4019bbc6:0x3ffee6f0 0x4019c185:0x3ffee710 0x400d7d8b:0x3ffee730 0x400d7de4:0x3ffee760 0x400d84c3:0x3ffee780 0x400d7b6d:0x3ffee7c0 0x400d7849:0x3ffee810 0x400d7871:0x3ffee850 0x400d5b41:0x3ffee8a0 0x400d60a9:0x3ffee970 0x400d616e:0x3ffeea20 0x400e48c1:0x3ffeea40 0x400e248a:0x3ffeea60
WARNING Found stack trace! Trying to decode it
[11:54:48]
[11:54:48]Rebooting...
[11:54:48]ets Jun  8 2016 00:22:57
[11:54:48]
[11:54:48]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:54:48]configsip: 0, SPIWP:0xee
[11:54:48]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:54:48]mode:DIO, clock div:2
[11:54:48]load:0x3fff0018,len:4
[11:54:48]load:0x3fff001c,len:952
[11:54:48]load:0x40078000,len:6084
[11:54:48]load:0x40080000,len:7944
[11:54:48]entry 0x40080310
```
Same chip and configuration.