esphome/esphome-core #260: TSL2561 error: Too much data to fit in transmitter buffer

derandiunddasbo, 2018-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** n/A

**ESP (ESP32/ESP8266/Board/Sonoff):** ESP32 / nodemcu-32s

**Affected component:** https://esphomelib.com/esphomeyaml/components/sensor/tsl2561.html

**Description of problem:**
After running fine for some months, a TSL2561 Ambient Light Sensor isn't initialized properly at reboot. The log shows an error `Too much data to fit in transmitter buffer for address 0x39`. I don't know, what this error means. Is the sensor recognized at all? Does it send wrong data? Or maybe is it just bad wiring?

Could somebody please enlighten me, how to interpret this error message? :-)

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 21
  scl: 22
  scan: False

sensor:
  - platform: tsl2561
    name: "EG Aquarium Helligkeit"
    address: 0x39
    update_interval: 15s

```

**Traceback (if applicable):**

```
[18:52:28][C][sensor.tsl2561:036]: Setting up TSL2561...
[18:52:28][C][sensor.tsl2561:037]:     Address: 0x39
[18:52:28][W][i2c:084]: Too much data to fit in transmitter buffer for address 0x39
[18:52:28][E][sensor.tsl2561:040]: Communication with TSL2561 on address 0x39 failed!
[18:52:28][E][component:162]: Component was marked as failed.
```

**Additional information:**

OttoWinter, 2018-11-08:
See note in https://esphomelib.com/esphomeyaml/components/i2c.html

derandiunddasbo, 2018-11-08:
With the `arduino_version: latest` option set, the error still occured after uploading. But I noticed, that the sensor recovers after power cycling the board.

I think I'll have to observe this behaviour. Maybe the sensor becomes unstable over time...

OttoWinter, 2018-11-09:
The error is coming directly from the i2c controller which is managed by the arduino core. As far as I know, esphomelib's i2c usage of this Arduino API is correct - and it is usually working quite well.

However, the i2c controller on the ESP32 is quite unstable and the arduino-esp32 guys have spent quite a bit of time working around the issue. Anyway, it's nothing esphomelib can fix. If it's a problem for you, I would recommend using ESP8266s for all devices using i2c, there the i2c controller is much more stable.

closed 2019-02-02 by OttoWinter

derandiunddasbo, 2019-05-07:
Just a little status update: About 4 month ago I replaced the ESP32 board for this sensor with a D1 Mini ESP8266-EX board and since then I didn't have any read errors from the sensor. As already supposed by @OttoWinter this was obviously an issue with the i2c controller on the ESP32 board I used before.