esphome/esphome-core #265: esphomeyaml automations without wifi or mqtt

snizzleorg, 2018-11-09:
I use esphomeyaml for an RFID door opener. recently home assistant and the mqtt broker had a problem and therefore my RFID door opener did not work. It would be great if the automations would still run even if no network connection is available. 

It seems now the system will loop and try to connect.

RolluS, 2018-11-11:
Hello,

Experienced a similar problem on sonoff relays:
- from automation, push button toggle the relays
- when mqqt server is not seen for a minute, relay goes down

I loose some WAF point when it happens to the washing machine...

OttoWinter, 2018-11-13:
esphomelib does actually run all automations while offline. One problem you might have experienced is the ESP *rebooting* because it can't establish a MQTT/WiFi connection. See the `reboot_timeout` options in [`mqtt:`](https://esphomelib.com/esphomeyaml/components/switch/remote_transmitter.html#raw-remote-codes) and [`wifi:`](https://esphomelib.com/esphomeyaml/components/wifi.html) where you can constumize this.

As for the RFID sensor, that's more of an architectural problem: When you add a component to esphomelib, by default it gets initialized after the MQTT client is fully set up. This is due to one reason: If there's an error with setting up the communication with the sensor, you'd want to have that logged in the *MQTT logs*. But if the sensor would get initialized before the MQTT client, the logs would only appear in the UART logs over USB.

I knew this would come some day, so I created https://github.com/OttoWinter/esphomelib/pull/267 to address this issue. With it, all components get set up before MQTT, but dump their configuration after everything (including MQTT logs) have been set up.

> when mqtt server is not seen for a minute, relay goes down

That's also a problem that has recently been fixed in the latest [beta](https://beta.esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-beta-release) -> esphomelib now stores the state of switches/lights/etc in the RTC memory so that they're restored on boot.

So that should fix all those issues :)

closed 2018-11-13 by OttoWinter

snizzleorg, 2018-11-13:
Sounds great. So the latest beta would work fully offline? Or does this Need any specific options?