esphome/esphome-core #269: Miflora with esp8266 + HM10/11 Bluetooth Module?

oakbrad, 2018-11-14:
**Is your feature request related to a problem? Please describe.**

Is it possible to use the current MiFlora implementation with an esp8266 and the HM-10/11 module? If not can support be added?

**Describe the solution you'd like**
I currently have two esp8266 nodes running OpenMQTTGateway with HM-11's as described here:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-bluetooth-BLE

It works wonderfully to detect both MiFlora sensor information and Bluetooth beacons.

These are the only esp8266 devices I haven't moved over to esphomelib :).

**Additional context**
I have about 20 MiFlora devices so if you are in need of hardware I would be happy to donate one :).

I have a couple of ESP32's - I flashed them with OpenMQTTGateway for the MiFloras but they are not stable - crash after a day or two. Have not tried the ESP32 with the esphomelib, but saw it was experimental. OpenMQTTGateway with the esp8266+HM11 is rock solid and never crashes.

OttoWinter, 2018-11-14:
Yes, that's being tracked here: https://github.com/OttoWinter/esphomeyaml/issues/77

I do own an HM10 module already. However, it seems to be a lot of work to implement a driver for that module so it's not at the top of my priority list - it's more about making the ESP32 integration more stable (which it did at least a bit in the latest beta https://beta.esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-beta-release)

oakbrad, 2018-11-15:
Thanks, didn't see that other issue. I have some ESP32's here so I'll report back on how that works.

closed 2018-11-15 by oakbrad