esphome/esphome-core #270: Template switch actions doesn't work.

voicevon, 2018-11-14:
Environment:  EsphomeLib V1.8.2 && V1.9.0.?  (Nov 14  2018)
                       EsphomeYaml V1.82
I reference this page https://esphomelib.com/esphomeyaml/components/switch/template.html, Yaml source code is:
```
  - platform: template
    name: "Template Switch"
    id: pump_switch
    lambda: >-
      if (id(leaking).value || id(lp).value) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - switch.turn_on:
          id: pump_A
    turn_off_action:
      - switch.turn_off:
          id: pump_A
    optimistic: true
```
Esphomeyaml created C++ code is: (I modified label names to read it better)
```
  Application::MakeTemplateSwitch application_maketemplateswitch = App.make_template_switch("Template Switch");
  application_maketemplateswitch.template_->set_state_lambda([=]() -> optional<bool> {
      if (lp->value) {
        return true;
      } else {
        return false;
      }
  });
  application_maketemplateswitch.template_->set_optimistic(false);

  Application::MakeGPIOSwitch application_makegpioswitch = App.make_gpio_switch("pump_A", GPIOOutputPin(PIN_PUMP_A, OUTPUT, true));
  switch_::Switch *pump_A = application_makegpioswitch.switch_;

  Automation<NoArg> *temp_turn_off_trigger = App.make_automation<NoArg>(application_maketemplateswitch.template_->get_turn_off_trigger());
  switch_::TurnOffAction<NoArg> *turn_off_acttion = pump_A->make_turn_off_action<NoArg>();
  temp_turn_off_trigger->add_actions({turn_off_acttion});

  Automation<NoArg> *temp_turn_on_trigger = App.make_automation<NoArg>(application_maketemplateswitch.template_->get_turn_on_trigger());
  switch_::TurnOnAction<NoArg> *turn_on_action = pump_A->make_turn_on_action<NoArg>();
  temp_turn_on_trigger->add_actions({turn_on_action});
```

Result:
1) The sensor works well. Can see sensor state from mqtt.
2) The template_switch lamda works well,  Can see this switch turn_on, turn_off
3) The pump_A switch has no response.

OttoWinter, 2018-11-14:
Unfortunately, you'll need to give me more information to help you. Can you show your `pump_A` configuration? - is it a GPIO switch? Does it have a `name`?

And what do you mean by "The pump_A switch has no response."? Is it only not showing up in hass or is it not working at all?

Also, just to be sure, can you try with https://beta.esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-beta-release

voicevon, 2018-11-16:
Yes, `pump_A `is a GPIO switch, it's the name and Id, same string. where `leaking` and `ld` is GPIO binary sensor.
I expect pump_A will be turned on when template_switch is turned on(depend on the sensors).
The result is when I pressed any of the two buttons. Mqtt monitor showed template_switch's state correctly, this proved template  lambda  works fine.      
    ```
lambda: >-
      if (id(leaking).value || id(lp).value) {
        return true;
      } else {
        return false;
      }
```
but the `pump_A` has no response,and I can't find any message from MQTT monitor about  `pumb_A` . this proved automation didn't work fine.
    ```
turn_on_action:
      - switch.turn_on:
          id: pump_A
    turn_off_action:
      - switch.turn_off:
          id: pump_A
    optimistic: true
```

I'am happy to see Version 1.9.0 includes a new feature of `Script` , So I can write a script to turn on/ turn off `pumb_A`. Meanwhile, I can remove my template_switch.  I will try soon.

OttoWinter, 2018-11-17:
Once again, without a [minimal working example](https://stackoverflow.com/help/mcve) I can't help you very much. I just tried to make such a template switch myself using the information you gave me, but that works correctly. So please give me a YAML file with which I can reproduce this, otherwise I don't know how I can help you here.

voicevon, 2018-11-27:
I reinstalled everything with the newest version. Those are Hassio, Esphomeyaml, Esphomelib, The problem has gone.Now it works well.  And, I am happy the `script` feature is very good for "share" repeated codes.