esphome/esphome-core #271: Add way to attach debug LED to sensors

Harvie, 2018-11-14:
Hello, i have PIR motion sensor attached to ESP32 but it does not have LED to quickly check it's state. I have defined LED output and i use nodered in homeassistant to turn the LED on when it's active. But i would rather have it directly and autonomously handled in ESP32, so i can easily tell that there's no problem in ESP or wifi when something goes wrong.

Just add something like led_indicator_pin option to each sensor, so i can attach LED pin to each sensor to show if it's active. Maybe with possibility to invert the state of that pin by entering negative number. 

Many ESP32 boards have builting LEDs, so we can use them for this by specifiing the pin. Also it would be cool if we can use single LED pin for multiple sensors (not sure how we should combine them, probably using AND operation).

OttoWinter, 2018-11-14:
> Just add something like led_indicator_pin option to each sensor

That's just wrong. If I would add such an option for each little use-case the base configuration for sensors would just get over-crowded.

Also, your use-case is easily handled by automations. Please read https://esphomelib.com/esphomeyaml/guides/automations.html - everything you need should be there already (using `binary_sensor.on_press` and `binary_sensor.on_release` triggers with `switch.turn_on` and `switch.turn_off` actions)

closed 2018-11-14 by OttoWinter

Harvie, 2018-11-14:
> https://esphomelib.com/esphomeyaml/guides/automations.html

Yes, that will probably work. (as far as it's handled directly on ESP)

> If I would add such an option for each little use-case the base configuration for sensors would just get over-crowded.

Maybe yes, maybe no. It's not really a usecase. It's more like debug case :-)

Maybe you can by default use builtin LED to indicate if any of the sensors is active. That way there's no  need to add anything to sensor options.

OttoWinter, 2018-11-14:
> Maybe you can by default use builtin LED to indicate if any of the sensors is active. That way there's no need to add anything to sensor options.

Well on many boards the LED pin is also connected to a user-facing GPIO pin - so writing to this pin could become an issue for many users.

Also, esphomelib has an important concept: Features are only activated if you manually put them in your config. This is done in order to a) reduce bloat b) minimize the firmware size and c) prevent "weird" behavior like a GPIO pin being written to without having manually told esphomelib to do so.

Again, if you want to have an indicator, it is super easy to do so yourself using automations.

Harvie, 2018-11-14:
> c) prevent "weird" behavior like a GPIO pin being written to without having manually told esphomelib to do so.

That's good

> Again, if you want to have an indicator, it is super easy to do so yourself using automations.

OK. thanks for your time.