esphome/esphome-core #272: filter_nan not working, nan values cause Home Assistant exception & break UI

oakbrad, 2018-11-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip on Linux

**ESP (ESP32/ESP8266/Board/Sonoff):**
All (ESP32, esp8266, sonoff)

**Affected component:**
Sensors - all types

**Description of problem:**
I have the filter_nan enabled for my sensors but nan values are still getting published to MQTT. These values will break the UI (I'm using Lovelace but can't get into the 'old' UI either) when they are displayed with Home Assistant error:

`2018-11-15 04:17:13 ERROR (MainThread) [frontend.js.latest.201810264] https://URL:8123/:1:237 Uncaught SyntaxError: Unexpected token N in JSON at position 236
2018-11-15 04:17:13 ERROR (MainThread) [frontend.js.latest.201810264] https:/URL:8123/:1:237 Uncaught SyntaxError: Unexpected token N in JSON at position 236
2018-11-15 04:17:13 ERROR (MainThread) [frontend.js.latest.201810264] https://URL:8123/dev-state:1:237 Uncaught SyntaxError: Unexpected token N in JSON at position 236`

Looking through the HA issues I came across this - some sort of type mismatch? https://github.com/home-assistant/home-assistant/issues/17547

If I look the retained topics on the broker I can see there are indeed a bunch with value nan:
ie:

```
home/BTtoMQTT/C47C8D6273BD/lux nan
home/BTtoMQTT/C47C8D6273BD/tem nan
home/BTtoMQTT/C47C8D6273BD/hum nan
home/BTtoMQTT/C47C8D627149/tem nan
home/BTtoMQTT/C47C8D627149/hum nan
home/BTtoMQTT/C47C8D627149/lux nan

```
If I clear them out manually with `mosquitto_pub -h mqtt.local -t bad/topic -n -r` and then restart HA, the UI comes back in Home Assistant. I can repeat this by adding a sensor with "nan" value in it's topic to the UI, and it will crash again.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:62:73:BD
    temperature:
      name: "Kitchen Pothos Temperature"
      state_topic: home/BTtoMQTT/C47C8D6273BD/tem
      discovery: false
      unit_of_measurement: "°F"
      filters:
        - filter_nan:
        - lambda: return x * (9.0/5.0) + 32.0;
    moisture:
      name: "Kitchen Pothos Moisture"
      state_topic: home/BTtoMQTT/C47C8D6273BD/hum
      discovery: false
      filters:
        - filter_nan:
    illuminance:
      name: "Kitchen Pothos Light Intensity"
      state_topic: home/BTtoMQTT/C47C8D6273BD/lux
      discovery: false
      filters:
        - filter_nan:
    battery_level:
      name: "Battery Kitchen Pothos"
      filters:
        - filter_nan:

```
Happens with all sensors on all my nodes, I just quoted the BLE ones as example here

OttoWinter, 2018-11-15:
With what esphomelib version is this. I believe this is fixed in the latest beta: https://beta.esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-beta-release

oakbrad, 2019-01-07:
I upgraded to the dev version and haven't seen the issue since. Thanks :)

closed 2019-01-07 by oakbrad