esphome/esphome-core #276: Setting HSV color via MQTT seems to have no effect

connyhald, 2018-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Just a simple RGB-Light project on ESP32 using MQTT. No other components involved for now.
Esphomelib v1.8.3

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP32

**Affected component:**

https://esphomelib.com/esphomeyaml/components/light/rgb.html

**Description of problem:**

After sending HS(B) values to esphomelib it reports back RGB-Colors that do not change and do not fit to the input HSB values. Also the color of the light does not change. If I send RGB values it works as expected.

```
mosquitto_pub -h 10.18.200.30 -t "home/rgblight01/command" -m '{"color":{"h":300.0, "s":50.0}, "brightness": 255}'
```

```
home/rgblight01/command {"color":{"h":300.0, "s":50.0}, "brightness": 255}
home/rgblight01/debug [D][light.state:024]: Starting transition with length=1000 ms
home/rgblight01/status {"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":127}}
home/rgblight01/debug [D][light.state:080]: Finished transformer.
home/rgblight01/status {"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":127}}
```

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2018-11-18:
HSV values are not supported - as far as I see the docs don't mention it either.

I would suggest you just convert them to RGB locally.

connyhald, 2018-11-18:
D'OH. I'm sorry, then it makes perfectly sense :smile: 

I think I was following some links bringing me to those pages and I didn't realize those are about Home Assistant.

https://www.home-assistant.io/components/light.mqtt_json/
https://github.com/home-assistant/home-assistant/pull/14029

Now looking at the URLs it obvious... Sorry again...

closed 2018-11-18 by connyhald

connyhald, 2018-11-18:
After re-reading the docs I recall how this misunderstanding happened.

I couldn't find out which MQTT/JSON messages are supported by esphomelibs RGB Light and the documentation on home-assistant.io looked promising, so I assumed that esphomelib implemented this protocol completely.