esphome/esphome-core #295: Wifi Troubles without debug Logger

Bierchermuesli, 2018-11-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
docker 1.9.2 

**ESP (ESP32/ESP8266/Board/Sonoff):**
nodemcuv2 (RobotDyn NodeMCU v3)
Tag: v1.9.0 / v1.9.1 / v1.9.2 
arduino_version: 2.4.2 / 2.4.1 / espressif8266@>=1.8.0

**Affected component:**
wifi / logger

**Description of problem:**
if log level is set to WARN, esp boots but wifi is not available / static IP not pingable

**Problem-relevant YAML-configuration entries:**
everything works: 
```
logger:
  # level: WARN
  level: DEBUG
debug:
```
Does not work (also if `debug:` is uncommented)
```
logger:
  level: WARN
  # level: DEBUG
debug:
```
**Traceback (if applicable):**
```
[W][wifi:128]: WiFi Connection lost... Reconnecting...
[W][wifi:365]: Timeout while connecting to WiFi.
...              
[W][wifi:365]: Timeout while connecting to WiFi.                        
[W][wifi:408]: Restarting WiFi adapter...                               
[W][mqtt.client:164]: MQTT Disconnected: WiFi disconnected.             
[W][wifi:365]: Timeout while connecting to WiFi.
...
[W][wifi:365]: Timeout while connecting to WiFi.
 

```

**Additional information:**
could be relate to #285 ?
not sure if the "no wifi"-behavior is only related to my kind of board.  (others will crash somehow?)

Bierchermuesli, 2018-11-26:
Dang... pretty much random. after the 23rd compiled image the logger thing mentioned above is not true anymore. I had to remove `fastled_clockless` component, wifi works fine since then...  

please ignore, I will check again. any Inputs are welcome :)

OttoWinter, 2018-11-27:
I don't think it's related to #285, that PR should only change the behavior when you completely remove the logger component, otherwise there's not really a change there.

The issue does look weird, and I think I saw someone with a similar issue some time ago (with `NONE` log level). But I still have no idea what would cause that.

OttoWinter, 2019-01-08:
Try with latest beta please

closed 2019-01-08 by OttoWinter

Bierchermuesli, 2019-01-16:
Looks like much better! Thanks! Also switched to api an dropped unnecessary sensors/http server etc... 
I guess I put too much on my nodemcu boards. Also, I got pretty cheap boards. 

I have >20oWifi Networks in my area, some of them with fancy SSID chars.. could this be an issue? I noticed the wifi crashed after the wifi scan (pretty huge list) - even if my network was listed/found.

OttoWinter, 2019-01-17:
> I have >20oWifi Networks in my area, some of them with fancy SSID chars.. could this be an issue? I noticed the wifi crashed after the wifi scan (pretty huge list) - even if my network was listed/found.

Interesting, this is 200 networks, right?

> fancy SSID chars

most likely that's not the issue here.

Do you see a "Connecting to ..." message after the WiFi list is printed or does it stop working even before that?