esphome/esphome-core #298: Vibration Sensor Module - SW-420

jjgadget, 2018-11-28:
**What new integration would you wish to have?**

I originally added this as a GPIO binary sensor but it triggers on/off on constantly.

**If possible, provide a link to an existing library for the integration:**
https://github.com/moopoint/vibration-pattern-sensor
This looks to possibly allow for the wait state between reporting.

**Is your feature request related to a problem? Please describe.**
To have this sensor supported so that it only reports on/off state after vibrations have started or stopped after X amount of seconds.

OttoWinter, 2018-11-28:
https://esphomelib.com/esphomeyaml/components/binary_sensor/index.html#binary-sensor-filters

OttoWinter, 2018-11-28:
No problem :)

closed 2018-11-28 by OttoWinter