esphome/esphome-core #302: Allow to set a factor for ADC measurements

download13, 2018-12-01:
The ESP8266 ADC only reads from 0-1v so we need a 3.2:1 voltage divider to bring the max 4.2v of a li-ion down to a readable range. The only problem is that the values that the ADC sensor component sends out are based on this 0-1v range so we aren't seeing the real battery voltage at the MQTT broker.

It would be cool if the ADC component had a `set_factor` method or something that would allow us to set a factor that the value is always multiplied by before being sent. That way for different battery chemistries/voltage divider configurations we can set the appropriate factor.

It's a simple enough change I could bash out a PR later if you're willing to add the feature (not sure how to make it settable through YAML though).

Thanks!

HFsi, 2018-12-02:
I think the sensor filters are what you asking for.
Inside the adc sensor you can add:
filters:
  - multiply: 3.2
to get the value you want

download13, 2018-12-02:
Whoops didn't notice those

closed 2018-12-02 by download13