esphome/esphome-core #308: BME680 reports bad pressure value

thomaskr, 2018-12-03:
# Description

Esphome yaml reports unreasonably low pressure value retrieved via esphomelib from Bosch BME680 sensor.

The pressure value is reported as 740 hPa while a value of 993 hPa is expected.

# Environment

esphomeyaml==1.9.3, Python 2.7.15rc1, Ubuntu 18.04 x86_64, [MASL](https://en.wikipedia.org/wiki/Metres_above_sea_level): 65

Dependency Graph
```
|-- <ArduinoOTA> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <Hash> v1.0
|-- <Wire> v1.0
|-- <esphomelib> v1.9.3 #9d5e290
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <ESPAsyncTCP> v1.2.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Hash> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <FastLED> v3.2.0
|   |   |-- <EspSoftwareSerial> v3.4.1
|   |-- <Wire> v1.0
|   |-- <ArduinoOTA> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
```

# Log output

```
[10:12:20][VV][component:132]: Running interval 'update':0 with interval=15000 last_execution=75656 (now=90663)
[10:12:20][VV][i2c:073]: Beginning Transmission to 0x77:
[10:12:20][VV][i2c:111]:     Writing 0b01110100 (0x74)
[10:12:20][VV][i2c:111]:     Writing 0b10100001 (0xA1)
[10:12:20][VV][i2c:078]:     Transmission ended. Status code: 0x00
[10:12:20][VV][component:079]: set_timeout(name='data', timeout=250)
[10:12:20][VV][component:132]: Running timeout 'data':1 with interval=250 last_execution=90684 (now=90950)
[10:12:20][VV][i2c:073]: Beginning Transmission to 0x77:
[10:12:20][VV][i2c:111]:     Writing 0b00011101 (0x1D)
[10:12:20][VV][i2c:078]:     Transmission ended. Status code: 0x00
[10:12:20][VV][i2c:100]: Requesting 15 bytes from 0x77:
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b01110101 (0x75)
[10:12:20][VV][i2c:132]:     Received 0b01111000 (0x78)
[10:12:20][VV][i2c:132]:     Received 0b11010000 (0xD0)
[10:12:20][VV][i2c:132]:     Received 0b01011111 (0x5F)
[10:12:20][VV][i2c:132]:     Received 0b01100010 (0x62)
[10:12:20][VV][i2c:132]:     Received 0b10000000 (0x80)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b00000000 (0x00)
[10:12:20][VV][i2c:132]:     Received 0b01001010 (0x4A)
[10:12:20][VV][i2c:132]:     Received 0b11111001 (0xF9)
[10:12:20][D][sensor.bme680:313]: Got temperature=19.5°C pressure=740.0hPa humidity=59.9% gas_resistance=18588.0Ω
[10:12:20][V][sensor.sensor:020]: 'Temperatur Flur Komode': Received new state 19.519823
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff3234)::input(19.519823)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff3234)::new_value(19.519823) -> 19.520716
[10:12:20][V][sensor.sensor:020]: 'Luftdruck': Received new state 739.966492
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff34fc)::input(739.966492)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff34fc)::new_value(739.966492) -> 739.967468
[10:12:20][V][sensor.sensor:020]: 'Luftfeuchtigkeit Flur': Received new state 59.895885
[10:12:20][VV][sensor.filter:022]: Filter(0x3fff37c4)::input(59.895885)
[10:12:20][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff37c4)::new_value(59.895885) -> 59.895073
[10:12:20][V][sensor.sensor:020]: 'Gassensor Flur': Received new state 18588.000000
[10:12:20][D][sensor.sensor:126]: 'Gassensor Flur': Sending state 18588.00000 with 0 decimals of accuracy
[10:12:20][V][mqtt.client:239]: Publish(topic='esp_hallway/sensor/gassensor_flur/state' payload='18588' retain=1)

```

mkihr, 2018-12-13:
I have exactly the same issue. Getting correct value when using Adafruit Lib for BME680.

pszafer, 2019-01-30:
Same bug for me...

BioGeekJoey, 2019-01-31:
Having the same issue, so it seems. 

A BME280 and a BME680 connected to the same D1 mini (esp8266) on the same I²C bus. Expected pressure is ~991.
![bme680][graphs]]

[graphs]:https://i.imgur.com/6QtP6Nk.png