esphome/esphome-core #311: Magic Byte Error

maxi1134, 2018-12-06:
Getting this error:

Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:0A:C4:B0:0D:34
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=65, should be E9)

Tried with 3 of those 
https://www.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32801337284.html

All give the same error:  Unexpected error: The firmware binary is invalid (magic byte=65, should be E9)

Please provide details about your environment.
-->  Windows with EshopmeFlasher, Yaml file generated on Docker

Please provide details about which ESP you're using.
--> 
this one https://www.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32801337284.html

**Description of problem:**

GEtting Unexpected error: The firmware binary is invalid (magic byte=65, should be E9)
 while trying to flash

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2018-12-07:
The error is correct, the firmware binary (the file you give the flasher) should always start with the bytecode `0xE9`. Could you please describe how you got the firmware file and what filename it has. I suspect you might be trying to upload the `firmware.elf` file, not the `firmware.bin` file.

maxi1134, 2018-12-07:
I'm an idiot. I was trying to upload the .Yaml.

Figured ish how to use the CLI, but now getting a similar error: 

 Invalid head of packet (0x1B)

```
  INFO Running:  platformio run -d bedroom -t upload --upload-port /dev/ttyUSB0
Processing bedroom (platform: espressif32@>=1.4.0; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphomelib> 1.9.3 #9d5e290
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <AsyncTCP> 1.0.1
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <AsyncTCP> 1.0.1
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Wire> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |-- <Preferences> 1.0
Retrieving maximum program size .pioenvs/bedroom/firmware.elf
Checking size .pioenvs/bedroom/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  13.6% (used 44576 bytes from 327680 bytes)
PROGRAM: [======    ]  59.1% (used 774894 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/bedroom/firmware.bin
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x1B)
*** [upload] Error 2
```

maxi1134, 2018-12-07:
Changed to a different Ubuntu machine and it worked. Problem solved

closed 2018-12-07 by maxi1134