esphome/esphome-core #312: How to customize FastLED effects

zouden, 2018-12-06:
I'm trying to customize the built-in rainbow effect by adjusting its width and speed. Here is the setup:

    fast_led.fast_led->add_leds<WS2812B, DATAPIN, GRB>(NUM_LEDS);
    fast_led.state->add_effects({new light::FastLEDRainbowLightEffect("Rainbow")});
    App.setup();

The rainbow effect works, but how do I edit the speed and width? According to the docs I should be able to use these functions:

    void set_speed(uint32_t speed)
    void set_width(uint32_t width)

But I'm unsure how to call them. Any advice would be much appreciated - I can tell Esphomelib is a great solution for what I want to do!

As a feature request, it would be nice to have the [fastled example sketch](https://github.com/OttoWinter/esphomelib/blob/master/examples/fastled/fastled.cpp) expanded to cover a few more cases like this.

OttoWinter, 2018-12-07:
[esphomeyaml](https://esphomelib.com/esphomeyaml/index.html) is the future of esphomelib - so the docs for esphomelib are not actively maintained apart from the docs that enable writing custom components. Similarly, the C++ API frequently has breaking changes (for example the code above will not work in 1.10.0), but the YAML syntax will not have a breaking change.

For your case, the way to do it would be to just use a variable:

```cpp
auto *effect = new light::FastLEDRainbowLightEffect("Rainbow");
effect->set_speed(42);
fast_led.state->add_effects({effect});
```

For anything else I would suggest looking at the output of the esphomeyaml command.

closed 2018-12-07 by OttoWinter

zouden, 2018-12-08:
Thanks Otto, that's exactly the syntax I needed. I'll switch over to Esphomeyaml soon though.