esphome/esphome-core #313: MAX44009 / GY-49 ambient light sensor support ?

vincegre, 2018-12-08:
Is it planned to get support or is there a workaround already for MAX44009 / GY-49 ambient light sensor ? It's an I2C module pretty efficient for light sensing.
Basic description of it here: https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.2bfd4c4ddKFYKF&orderId=507448985225914&productId=32804395737

Thanks

Vincèn

Esiravegna, 2018-12-10:
I do already have these implemented as a custom sensor, if it'd may help, I can share the code with you guys.

vincegre, 2018-12-12:
> I do already have these implemented as a custom sensor, if it'd may help, I can share the code with you guys.

yep please it would be really nice ;)

Esiravegna, 2018-12-14:
There you go:
```
#include "esphomelib/application.h"
#include <Max44009.h>

using namespace esphomelib;

class MAX44009Sensor : public sensor::PollingSensorComponent {
 public:
    Max44009 tsl = Max44009(0x4A, 21, 22); //these are the i2c pins being used

  MAX44009Sensor(const std::string &name, uint32_t update_interval) : sensor::PollingSensorComponent(
    name, update_interval) {}
  
  void update() override {
    float lux = tsl.getLux(); // in lux
    int err = tsl.getError();
    if(err) 
      ESP_LOGI("luxmeter", "Got error code: %1i", err);  
    else   
      //ESP_LOGD(TAG, "Got lux=%.1flux", lux);   
      push_new_value(lux); 
  }

   std::string icon() override {
    return "mdi:theme-light-dark";
  }

    std::string unit_of_measurement() override { return "Lux"; }
  int8_t accuracy_decimals() override { return 2; } // 2 decimal places of accuracy.
};

```

And this in your setup method after the markers:

```
 auto *luxmeter = App.register_component( new MAX44009Sensor("Light Level", 30 * 1000)); 
 App.register_sensor(luxmeter);
 
```

vincegre, 2018-12-14:
> There you go:

Thanks for the share ;) Where should I put that code to be able to compile it in esphomelib ? and how did you declare in configuration file ?

Esiravegna, 2018-12-14:
The configuration file is used to generate code, hardly the custom sensors would use any YAML.  Please refer to this [guide](https://esphomelib.com/esphomeyaml/components/sensor/custom.html) in order to use it :).