esphome/esphome-core #314: Min - Max values for sensor (Rotary Encoder)

vash3d, 2018-12-10:
It would be very usefull to have the possibility to add a MIN and MAX value for a sensor like rotary encoder.
Something like it is used in ESPEasy:
Limit min: is the lowest value that the counter will go to. Observe that if you have the counter on a value below this limit the counter will climb above this value but not below it once it has reached the min limit.

Limit max: is the highest value that the counter will go to. As with the min limit if the counter is above this limit it will descend to the max level but never climb over it once reached below it.

vash3d, 2018-12-10:
I'm not a C++ programmer but looking at the code, could this be some sort of implementation?

```
...
if (counter_change && counter_ >= max_value) {         // max_value variable setted elsewhere, as a filter for the sensor for example
      this->counter_ = max_value;
    }
...
```

OttoWinter, 2018-12-11:
That would be a task for sensor filters. See https://community.home-assistant.io/t/esphomelib-library-to-greatly-simplify-home-assistant-integration-with-esp32/40245/195?u=ottowinter

closed 2018-12-11 by OttoWinter