esphome/esphome-core #315: mqtt subsript sensor doesn't work on esp8266

voicevon, 2018-12-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266/Board/Sonoff):**
Esp8266

**Affected component:**
https://esphomelib.com/esphomeyaml/components/sensor/mqtt_subscribe.html
After resetting,The node got the correct value on the first time, but when I  publish a new value, the esp node has no response, I even tried send a non-number with `"abc"`, the debugger didn't show any warning information.

**Problem-relevant YAML-configuration entries:**
```sensor:
  - platform: mqtt_subscribe
    name: "starter"
    id: starter
    topic: feeder/sensor/starter/state

```

**Traceback (if applicable):**

```
␛[0;35m[C][ota:472]: There have been 2 suspected unsuccessful boot attempts.␛[0m
␛[0;32m[I][application:053]: Running through setup()...␛[0m
␛[0;35m[C][switch.gpio:023]: Setting up GPIO Switch 'motor'...␛[0m
␛[0;36m[D][switch.gpio:026]:   Restored state OFF␛[0m
␛[0;36m[D][switch:037]: 'motor' Turning OFF.␛[0m
␛[0;36m[D][switch:055]: 'motor': Sending state OFF␛[0m
␛[0;35m[C][wifi:025]: Setting up WiFi...␛[0m
␛[0;32m[I][wifi:277]: WiFi Connecting to 'FuckGFW'...␛[0m
␛[0;32m[I][wifi:345]: WiFi connected!␛[0m
␛[0;35m[C][wifi:323]:   SSID: 'FuckGFW'␛[0m
␛[0;35m[C][wifi:324]:   IP Address: 192.168.123.239␛[0m
␛[0;35m[C][wifi:326]:   BSSID: CC:34:29:F9:53:BA␛[0m
␛[0;35m[C][wifi:328]:   Hostname: 'feeder'␛[0m
␛[0;35m[C][wifi:330]:   Channel: 5␛[0m
␛[0;35m[C][wifi:331]:   Subnet: 255.255.255.0␛[0m
␛[0;35m[C][wifi:332]:   Gateway: 192.168.123.1␛[0m
␛[0;35m[C][wifi:333]:   DNS1: 192.168.123.1␛[0m
␛[0;35m[C][wifi:334]:   DNS2: 0.0.0.0␛[0m
␛[0;35m[C][ota:140]: Over-The-Air Updates:␛[0m
␛[0;35m[C][ota:141]:   Address: 192.168.123.239:8266␛[0m
␛[0;35m[C][ota:143]:   Using Password.␛[0m
␛[0;33m[W][ota:147]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts␛[0m
␛[0;35m[C][mqtt.client:024]: Setting up MQTT...␛[0m
␛[0;36m[D][mqtt.client:111]: Resolving MQTT broker IP address...␛[0m
␛[0;36m[D][mqtt.client:144]: Resolved broker IP address to 123.233.2.137␛[0m
␛[0;32m[I][mqtt.client:170]: Connecting to MQTT...␛[0m
␛[0;32m[I][mqtt.client:209]: MQTT Connected!␛[0m
␛[0;32m[I][application:089]: setup() finished successfully!␛[0m
␛[0;32m[I][application:094]: You're running esphomelib v1.10.0-dev compiled on Dec 10 2018, 19:23:13␛[0m
␛[0;35m[C][wifi:338]: WiFi:␛[0m
␛[0;35m[C][wifi:323]:   SSID: 'FuckGFW'␛[0m
␛[0;35m[C][wifi:324]:   IP Address: 192.168.123.239␛[0m
␛[0;35m[C][wifi:326]:   BSSID: CC:34:29:F9:53:BA␛[0m
␛[0;35m[C][wifi:328]:   Hostname: 'feeder'␛[0m
␛[0;35m[C][wifi:330]:   Channel: 5␛[0m
␛[0;35m[C][wifi:331]:   Subnet: 255.255.255.0␛[0m
␛[0;35m[C][wifi:332]:   Gateway: 192.168.123.1␛[0m
␛[0;35m[C][wifi:333]:   DNS1: 192.168.123.1␛[0m
␛[0;35m[C][wifi:334]:   DNS2: 0.0.0.0␛[0m
␛[0;35m[C][switch.gpio:034]: GPIO Switch 'motor':␛[0m
␛[0;35m[C][switch.gpio:035]:   Pin: GPIO16 (Mode: OUTPUT)␛[0m
␛[0;35m[C][logger:094]: Logger:␛[0m
␛[0;35m[C][logger:095]:   Level: DEBUG␛[0m
␛[0;35m[C][logger:096]:   Log Baud Rate: 115200␛[0m
␛[0;35m[C][ota:140]: Over-The-Air Updates:␛[0m
␛[0;35m[C][ota:141]:   Address: 192.168.123.239:8266␛[0m
␛[0;35m[C][ota:143]:   Using Password.␛[0m
␛[0;33m[W][ota:147]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts␛[0m
␛[0;35m[C][mqtt.client:058]: MQTT:␛[0m
␛[0;35m[C][mqtt.client:060]:   Server Address: voicevon.vicp.io:1883 (123.233.2.137)␛[0m
␛[0;35m[C][mqtt.client:061]:   Username: 'von'␛[0m
␛[0;35m[C][mqtt.client:062]:   Client ID: 'feeder-6001941006e2'␛[0m
␛[0;35m[C][mqtt.client:064]:   Discovery prefix: 'homeassistant'␛[0m
␛[0;35m[C][mqtt.client:065]:   Discovery retain: YES␛[0m
␛[0;35m[C][mqtt.client:067]:   Topic Prefix: 'feeder'␛[0m
␛[0;35m[C][mqtt.client:069]:   Log Topic: 'feeder/debug'␛[0m
␛[0;35m[C][mqtt.client:072]:   Availability: 'feeder/status'␛[0m
␛[0;35m[C][sensor.mqtt_subscribe:042]: MQTT Subscribe Sensor 'starter':␛[0m
␛[0;35m[C][sensor.mqtt_subscribe:043]:   Topic: feeder/sensor/starter/state␛[0m
␛[0;35m[C][sensor.mqtt:029]: MQTT Sensor 'starter':␛[0m
␛[0;35m[C][sensor.mqtt:033]:   Unit of Measurement: ''␛[0m
␛[0;35m[C][sensor.mqtt:034]:   Accuracy Decimals: 0␛[0m
␛[0;35m[C][sensor.mqtt:041]:   State Topic: 'feeder/sensor/starter/state'␛[0m
␛[0;35m[C][switch.mqtt:048]: MQTT switch 'motor': ␛[0m
␛[0;35m[C][switch.mqtt:055]:   State Topic: 'feeder/switch/motor/state'␛[0m
␛[0;35m[C][switch.mqtt:055]:   Command Topic: 'feeder/switch/motor/command'␛[0m
␛[0;32m[I][application:108]: Running through first loop()␛[0m
␛[0;36m[D][sensor.sensor:131]: 'starter': Sending state 9.00000 with 0 decimals of accuracy␛[0m
␛[0;32m[I][application:135]: First loop finished successfully!␛[0m
␛[0;32m[I][ota:165]: Boot seems successful, resetting boot loop counter.␛[0m

```

**Additional information:**
My other is my mobile phone mqtt-dash app that it subscript  the same topic. it shows the sensor with the new value.

HFsi, 2018-12-10:
If it´s just for turning on and off something, I use the on_message trigger in the mqtt section.
Then I send an unretained message with just the topic, eg: /d1mini/luz_quincho
And the on_message automation toggles a switch.
Works great for me

voicevon, 2018-12-11:
Thank you @HFsi .  I am using the switch to verity the mqtt_subscribe_sensor. 
Actually, I just want to get an number from somewhere(like mqtt dash), This number  determines how long time my motor will keep running. I tried your method, I am sure that my target number is inside the payload of that message,   but it's "Hidden",  I don't know how to get that number from `on_message` trigger.

OttoWinter, 2018-12-11:
I believe the issue might be connected to failed subscriptions. Try with the latest [edge](https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version) version.

voicevon, 2018-12-11:
I am sure the connection to mqtt broke has no problem, because another binary_gpio_sensor is working.
Since I am using vs-code + Platformio, I can sync the esphomelib source code and compile it. I want to know whether the edge branch (the default branch under OttoWinter/esphomelib) is exactly as same as [edge](https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version) (need to uninstall and reinstall the addon).

voicevon, 2018-12-26:
Today, I reinsatalled Hassio + Esphomeyaml edge.The new UI and uploading function works great! But unfortunately , The mqtt_on_message is not good yet. Here is my code:
`esphomeyaml:
  name: feeder
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: 'FuckGFW'
  password: 'refuckgfw'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '1234567890'

ota:
  password: '1234567890'
  
  
switch:
  - platform: gpio
    pin: 5
    id: motor_p1
    internal: false
    
  - platform: gpio
    pin: 4
    id: motor_p2

status_led:
  pin: 
    number: 2
    inverted: true
  
  
mqtt:
  broker: 'xxxx'
  username: 'xxxx'
  password: 'xxxx'
  id: mqtt
  on_message: 
    - topic: feeder/switch/motor/command
      then:
        - lambda: |
            id(motor_p1).turn_on();
`
1) Complied, and uploaded without any problem, But no response when I send message.
2)  I copied all code to vs-code, comment these two lines  
```
  // motor_p1->set_internal(true);
  // motor_p2->set_internal(true);
```
3)The GPIO_switch works very well.
4) I repeated step1, try to find information from vs-code terminal window. But nothing about "mqtt_on_message" I am sure:
4a) the topic string has no problem.
4b) I tried both retained message and unretained message. 

I don't know whether  this is the issue only on Esp8266 or not. @HFsi , Were you using Esp8266 or Esp32?

HFsi, 2018-12-26:
I use ESP8266.
How do you send the MQTT message?
In my case, the topic that controls the trigger isn't correlated to any id or internal config, eg:

 on_message:
    - topic: sonoff_dual/luz
      then:
       - script.execute: scr_luz

Where "luz" is nothing, simply a word to activate de below script.
I don't know if I'm clear enough

voicevon, 2018-12-26:
@HFsi  Thank you very much.
I tried your way.
```
mqtt:
  broker: 'xxxx'
  username: 'xxxx'
  password: 'xxxx'
  on_message: 
    - topic: abcd/efgh
      then:
        - script.execute: qwert

script:
 - id: qwert
   then:
    - switch.turn_on: motor_p1
```
 but has no affect. 
The second trying is removing script.  
```
  on_message: 
    - topic: abcd/efgh
      then:
      - switch.turn_on: motor_p1
```
Still no response.

I sent message with 9(maybe more) styles those are combined with Qos{0/1/2} + retained/none-retained + payload={empty/ number/string}.

It seems the difference between you and me is (or might be)
1) The board. NodeMCU vs Snooff.    They are so similar 
2) Another potential thing is mqtt server I am using " RespB3 plus“  +  ”mosquito“. I don't no about your side.
3) I have two RespPi,Same hardware. First  is Hassio+mqtt. The second server is Hassio + Esphomeyaml.

voicevon, 2018-12-26:
@HFsi , Did you see any debug information about on_message's setup or received?

HFsi, 2018-12-26:
This is how I create the message for the topic, in scripts.yaml

toogle_luz_quincho:
  sequence:
  - service: mqtt.publish
    data: {"topic":"sonoff_dual/luz","retain":"false"}

Maybe can help

voicevon, 2018-12-27:
Thank you @HFsi . But I couldn't understand that.  (My background is C programmer)
I debugged c++ source code, I found the reason closely right now. I hope the below can help @OttoWinter  

```
//This is invoked before mqtt_connected,
void MQTTClientComponent::subscribe(const std::string &topic, mqtt_callback_t callback, uint8_t qos) {
  MQTTSubscription subscription{
      .topic = topic,
      .qos = qos,
      .callback = std::move(callback),
      .subscribed = false,
      .resubscribe_timeout = 0,
  };  Serial.println("=====================  MQTTClientComponent::subscribe()" );
  this->resubscribe_subscription_(&subscription);
  this->subscriptions_.push_back(subscription);
}

//this is invoked after mqtt_connected
void MQTTClientComponent::resubscribe_subscriptions_() {
  Serial.printf("---------------------Subscriptions size = %i",this->subscriptions_.size());
  Serial.println();
  for (size_t i = 0; i < this->subscriptions_.size(); i++) {
    
   Serial.printf("------------------------------------ %i",i);
    this->resubscribe_subscription_(&this->subscriptions_[i]);
  }
  Serial.println();
}
```
1)In main.cpp.  there is two gpio_switch , and one mqtt_on_message_trigger. 
1A) Before mqtt_connected, I saw two subscription ,it is motor_p1 and motor_p2 I guess. But It should be 3 subscriptions.  
1B) After mqtt_connected,   subscription size =2
2) In main.cpp,  I removed all gpio_switch, also removed on_message_trigger.
2A)   Before mqtt_connected, No subscription.
2B) After mqtt_connected, subscription size =0
3) Conclusion:
3a) The on_message topic has not been subscript  to broker.
3b) Mqtt_onmessage is using other way to subscribe a topic. 
4) Some thought
4a)In main.cpp setup(), add this line:  App.get_mqtt_client()->subscribe("abcd/efg", callback_function ,0);
I don't know the code of callback_function.

closed 2018-12-28 by voicevon