esphome/esphome-core #320: logger: VERBOSE causes 'SYSTEM_EVENT_AP_STAIPASSIGNED' was not declared in this scope

tringler, 2018-12-13:
Hello,

I have a strange behaviour with `logger:` - If I set it to `VERBOSE` with that configuration I get `.piolibdeps/esphomelib/src/esphomelib/wifi_component.cpp:183:10: error: 'SYSTEM_EVENT_AP_STAIPASSIGNED' was not declared in this scope`

```
esphomeyaml:
  name: floor
  platform: ESP32
  board: nodemcu-32s
  arduino_version: espressif32@1.0.2

wifi:
  ssid: TRUDI
  password: erdfcv!qwasyx!erdfcv

logger:
  level: VERBOSE
```
If I remove `level: VERBOSE` it compiles successfully.

OttoWinter, 2018-12-15:
`espressif32@1.0.2` is not the latest arduino version (and only recent ones are supported officially). Try removing that line

closed 2018-12-15 by OttoWinter

tringler, 2018-12-15:
As far as I know I need the `espressif32@1.0.2` line to get my bluetooth module on my ESP32 running